raspberrypi/userland #343: How to decode MJPEG from raspivid

AlexandreLicinio, 2016-10-06:
Hi, 
We tried to decode raspivid -w 1280 -h 720 -fps 25 -cd MJPEG -t 0 -o - through ffmpeg, mpv, vlc and gstreamer but not possible. Nothing on the web about it.
Do you people tried to decode MJPEG stream from raspivid with gstreamer, hellovideo_bin or anything else ?
we can reach an 115ms end-to-end latency in h264 but need to decrease it so we did think about MJPEG option in raspivid.

luked99, 2016-10-06:
It seems to work for me, at least if I play it via a file through mplayer on an x86 linux machine. Mplayer reports:

    VIDEO:  [MJPG]  1280x720  0bpp  25.000 fps    0.0 kbps ( 0.0 kbyte/s)

6by9, 2016-10-06:
You may need to multiplex it into a container first.
Certainly VLC used (around 2014) to not support an MJPEG Elementary Stream (ie the raw codec bits) as it took it as a standard JPEG, but would take MJPEG muxed into eg MP4 or MKV.  I think more recent versions handle it better now as I don't remember it being a problem last time I tried, and luked99 isn't seeing a problem now.

Google foo: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=62364&start=75#p473486
> Good news for those after MJPEG - I have something working that is producing a nice stream of JPEGs. VLC doesn't like it (just displays the first frame, but does correctly get the clip duration as 10s long), but omxplayer is quite happy. v4l2-ctl claims to be getting 30fps at 1920x1080 from it too.

AlexandreLicinio, 2016-10-06:
Thanks, both of you, i was quite sure using a container.
Do you think we can reach a lower latency with MJPEG than we have in H264 (115ms) ?

How should i proceed to pass a container for streaming ? and wich one ? i tried with ffmpeg, i can produce a file readable but not a stream... 
Because something like ffmpeg -i -codec copy  -f mpegts udp://localhost:5000 doesn't work, i mean vlc, mpv are not happy

6by9, 2016-10-06:
Just basing the numbers on the hardware being specified for 1080P30, 720P is x2.25 fewer pixels, so a linear conversion would say 67.5fps or 14ms/frame.

It's not quite that simple though as the H264 encode pipe is pipelined - CABAC is done a separate stage. Total encode of a 1080P frame takes about 40ms - ~32ms for the encode, and ~8ms for the CABAC compression. 
So we're looking at probably about 18ms for a 720P frame encode based on the same scaling. Would MJPEG improve on that? Possibly by a couple of ms, but you'll need a much higher bitrate to carry the same image quality, therefore increasing your network latency.

Break the problem down and take measurements (I've been through this loop before - https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=153410&p=1027417 looks like one such thread, or just google for "site:raspberrypi.org MMAL_PARAMETER_SYSTEM_TIME" for multiple hits).

Every frame coming from the GPU has a timestamp which is the GPU clock at the frame start interrupt on the camera interface. Userspace can also ask what the current GPU clock reads. When you get the buffer callback then get the current time, subtract the timestamp, and that is the processing time the GPU took over producing the frame. It does NOT include the exposure time on the sensor, which will typically be up to 33ms unless limited by the frame rate.
Now switch between H264 and MJPEG. What is the difference? Is it significant enough to you to continue investigations?

You may be able to save a couple of milliseconds by disabling CABAC (again image quality will go down for he same bitrate).
If you're always under reasonable lighting conditions then you may save the most by reducing the exposure time - easiest way is to choose the "sport" exposure mode to prefer gain (and noise) in preference to exposure time. That won't show up in the latency I've mentioned above, as that is the time before the frame start interrupt.

luked99, 2016-10-06:
> How should i proceed to pass a container for streaming ? and wich one ? i tried with ffmpeg, i can produce a file readable but not a stream... 

I found this suggestion. I haven't tried it myself:

http://raspberrypi.stackexchange.com/questions/27082/how-to-stream-raspivid-to-linux-and-osx-using-gstreamer-vlc-or-netcat

luked99, 2016-10-06:
> Do you think we can reach a lower latency with MJPEG than we have in H264 (115ms) ?

Striped mode was talked about a long time ago; the camera is generating stripes anyway (unfortunately only in stills mode, probably not in video mode) and the encoder should then be able to encode them one stripe at a time. That might help as your stripes would be "newer" on average. 6by9 will be along in a moment to tell me this is not possible though.

AlexandreLicinio, 2016-10-06:
6by9, every post you made are gold my dear.
Thanks to bright my mind :)
Luked99 i will read the post, very interesting, thanks you too

6by9, 2016-10-06:
Striped mode was only used for Miracast (or demo) IIRC, certainly it's not in any of the main code lines that I can recall. Switching to striped mode also restricts the software processing stages that can be active.
I can't remember what they did in the codec over the CME in striped mode. I thought it only had a full frame "go" activation, so how did that get throttled?

Basic answer is I'm not intending on investigating passing stripes of image from camera to video_encode - it'll be a very marginal gain, have many restrictions, and other tasks have been waiting around for much longer. 

Dom and Gert did a low latency streaming solution for one of Gert's demos. http://www.gertbot.com/toaste.html
I don't know what numbers they got, but Dom will probably have pulled out most the stops on the decode side. I don't know whether the apps have bit-rotted or not. I can't see source based on a quick glance :-(

popcornmix, 2016-10-06:
See: https://www.raspberrypi.org/forums/viewtopic.php?t=93330
but it's about 200ms end to end.

6by9, 2016-10-06:
Thanks for the link popcornmix. I wasn't sure what level you'd achieved.

33ms exposure time.
~30ms to complete the frame processing
~20ms for H264 encode.
?? transmission latency
~10ms to decode the frame on a Pi.
16ms if 60Hz HDMI refresh (worst case?).
= 109ms ignoring transmission latency.
None of those can easily be significantly reduced based on the hardware. The exposure time is the largest component, so minimising that is probably the biggest gain.

AlexandreLicinio, 2016-10-06:
Is there any intra-refresh (true intra-refresh) mode in raspivid ? where we can send only I image, i did reduce GOP to 1 but nothing big in gain's latency. 
115ms is quite nice but we want to reach 60ms like analog wireless system.

6by9, 2016-10-06:
Do you mean periodic intra refresh? If so, then no it is not supported.

Why do you think switching to I-frames only would improve latency significantly? It saves the motion estimation stage, but that's a fraction of the encode time. Almost certainly transmitting the extra bits will take about the same amount of time as the motion estimation.

Simple answer is that you're unlikely to be able to achieve 60ms with any form of digital compression. Processing pixels always takes time. Exposure time is almost always going to be the biggest factor - address that first (sports mode).
Drop the resolution and you'll make some savings. Switch to camera mode 6 (VGA 30-60 fps) and you'll get each frame in and processed in ~15ms, plus ~6ms for H264 encode. That'll save you around 30ms.
I don't know if you can increase the HDMI refresh rate on your display (again dropping the resolution). That would reduce the display latency, and having reduced the encoding resolution above you'll also save some time on the actual decode.

AlexandreLicinio, 2016-10-07:
Yes periodic, like 422.10bits broadcast encoder.

So yes u're right sports mode is very nice, i think we saved again 5ms.
And yes VGA mode decrease again latency, but we want 720p at least.

We can reduce latency but moving in jpeg2000 and fpga board an we will reach a minimal latency like 25ms.

thanks 6by9

6by9, 2016-10-07:
And how much are you paying for that YUV422 10bit broadcast encoder? Rather more than $35!

AlexandreLicinio, 2016-10-07:
yes, it is a custom made for the tour de france that we use in motorbike and helicopters, 60ms, propriatary protocol, OFDM links. each encoder is like 30k€, raaaaaather more than 35$ !

AlexandreLicinio, 2016-10-07:
i tried omxplayer, works fine but just for 5sec.. do you know why ?

 raspivid -w 840 -h 480 -fps 15 -cd MJPEG -t 0 -o - > mjpeg.h264 &
 omxplayer mjpeg.h264 

omxplayer decodes well, then freeze

6by9, 2016-10-07:
"mjpeg.h264" - well that's an interesting combination.

At a guess omxplayer has hit the end of the file and stopped. You haven't given it a frame rate so it'll play as fast as it can.

The MJPEG codec also isn't as robust as H264. There is a known issue that if you don't consume the output data fast enough then it may wedge on shutdown. If you're using the camera, set MMAL_PARAMETER_CAPTURE to 0 on the camera capture port, pause briefly whilst still consuming output frames from video_encode, then shut down the codec and you should be OK. You should actually get a frame with the EOS flag set on it through from video_encode too, and that way you know the pipe has emptied.
Probably ought to add that to raspivid, although many people will just hit Ctrl-C, so need to add signal handlers too.

AlexandreLicinio, 2016-10-07:
Yes, i just wanted to put in .something !

I'm not good in C, do you know where i can find a tutorial where i can setup that ?
Where should i change MMAL_PARAMETER_CAPTURE ?

latency seems VERY LOW !
thanks

AlexandreLicinio, 2016-10-07:
ah it works ! 

i just did ctrl-c on raspivid to relaunch it but while omxplayer was crashed, then omxplayer did restart automatically and mark :
V:PortSettingsChanged: 840x480@25.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

yes i did launch raspivid with raspivid -w 840 -h 480 -fps 15 -cd MJPEG -t 0 -o - > mjpeg.h264
so very good ! and now it is stable

6by9, 2016-10-07:
> There is a known issue that if you don't consume the output data fast enough then it may wedge on shutdown.

AlexandreLicinio, 2016-10-09:
I will close the issue because i have now enough informations to go further.
Thanks 6by9, thanks popcornmix and luked99

closed 2016-10-09 by AlexandreLicinio