raspberrypi/userland #348: DISPMANX_MODEINFO_T transform value sometimes correct

waveform80, 2016-10-20:
I've been working on adding screenshot capabilities to picamera and ran across something a little odd in the result of `vc_dispmanx_display_get_info`. When `display_rotate` (in `/boot/config.txt`) is set to 0, 1, or 3 (no rotation, 90 degrees, or 270 degrees), the info returned by `vc_dispmanx_display_get_info` is as expected: `width` and `height` are all correct (swapped in the case of 90 or 270 degree rotation), and the `transform` field reports the display rotation (0, 1, or 3).

However, when `display_rotate` is set to `2` (180 degree rotation), `width` and `height` are correct (same as unrotated), but `transform` reads `0` (the display is definitely rotated though). The same happens for 65536 and 131072 (horizontal and vertical flip) too, but I'm not sure if those are documented use-cases.

The following little snippet of C should demonstrate things:

```c
#include <stdio.h>
#include "bcm_host.h"

int main(void) {
    DISPMANX_DISPLAY_HANDLE_T display;
    DISPMANX_MODEINFO_T info;

    bcm_host_init();
    display = vc_dispmanx_display_open(0);
    vc_dispmanx_display_get_info(display, &info);
    printf("Size: %dx%d, Transform: %d\n", info.width, info.height, info.transform);
}
```

popcornmix, 2016-10-20:
Are you happy just getting the value of display_rotate back?
i.e. 0, 1, 2, 3 for standard rotations with 65536 or 131072 added if there are flips?

waveform80, 2016-10-20:
Yup, that'd be perfect.

I've had a look at [raspi2png's source code](https://github.com/AndrewFromMelbourne/raspi2png/blob/master/raspi2png.c) while I'm at it and it seems to work around the issue by using `vc_gencmd` to query the `display_rotate` parameter directly. I haven't translated that header yet, so this getting fixed might save me a bit of work (and from the looks of it, vc_gencmd involves some string parsing under the covers - always nice to avoid stuff like that if possible :).

popcornmix, 2016-10-23:
Test firmware you can try: https://www.dropbox.com/s/oo5f4ztfozgeg06/firmware_ip.zip?dl=0

waveform80, 2016-10-23:
Many thanks - that works nicely!

JamesH65, 2017-12-05:
@popcornmix Did this get merged?

popcornmix, 2017-12-05:
Yes (you can see the firmware updates that link to this issue above)

JamesH65, 2017-12-05:
Cool, closing.

closed 2017-12-05 by JamesH65