raspberrypi/userland #356: Valid transforms for vc_dispmanx_snapshot?

waveform80, 2016-11-14:
This is a bit of a follow-on from #348. I'm trying to figure out what transforms can be applied using the `vc_dispmanx_snapshot` function. The [comment above its declaration](https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vc_dispmanx.h#L127) states only transform 0 is supported but this doesn't seem to be *entirely* true.

The [declaration of `DISPMANX_TRANSFORM_T`](https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vc_dispmanx_types.h#L73) has a chunk at the end commented as "extra flags for controlling snapshot behaviour". That suggests that at least those transforms are probably supported for snapshot and a few experiments have born this out (swap red/blue seems to work nicely for example - I haven't quite figured out what "no-rgb" and "no-yuv" are about yet though!). Anyway, the fact these are tagged on the end of the transform enum rather suggested to me that other transforms might be implemented too (otherwise why wasn't an entirely separate enum specifically for snapshot transforms used?).

The results of experimenting with them are ... a little odd. For example, here's a snippet that might save an unencoded RGB snapshot of the display rotated by 180 degrees:

```c
#include <stdio.h>
#include "bcm_host.h"

int main(void) {
    DISPMANX_DISPLAY_HANDLE_T display;
    DISPMANX_RESOURCE_HANDLE_T resource;
    DISPMANX_MODEINFO_T info;
    VC_RECT_T rect;
    int res_width, res_height, pitch, image, rc;
    void *image_data;
    FILE *output;

    bcm_host_init();
    display = vc_dispmanx_display_open(0);
    if (!display) {
        fprintf(stderr, "Failed to open display 0\n");
        goto error;
    }
    vc_dispmanx_display_get_info(display, &info);
    /* Transpose size if system configuration applies a 90/270 rotation */
    if (info.transform & 0x1) {
        res_width = info.height;
        res_height = info.width;
    }
    else {
        res_width = info.width;
        res_height = info.height;
    }
    pitch = 3 * VCOS_ALIGN_UP(res_width, 16);
    image_data = malloc(pitch * res_height);
    if (!image_data) {
        fprintf(stderr, "Failed to allocate space for image data\n");
        goto error;
    }
    resource = vc_dispmanx_resource_create(
        VC_IMAGE_RGB888, res_width, res_height, &image);
    if (!resource) {
        fprintf(stderr, "Failed to create resource for snapshot\n");
        goto error;
    }
    /* Try getting snapshot to rotate its output... */
    if (vc_dispmanx_snapshot(display, resource, DISPMANX_ROTATE_180)) {
        fprintf(stderr, "Failed to capture snapshot\n");
        goto error;
    }
    if (vc_dispmanx_rect_set(&rect, 0, 0, res_width, res_height)) {
        fprintf(stderr, "Failed to setup rectangle\n");
        goto error;
    }
    if (vc_dispmanx_resource_read_data(resource, &rect, image_data, pitch)) {
        fprintf(stderr, "Failed to copy snapshot to resource\n");
        goto error;
    }
    output = fopen("image.data", "w");
    if (!output) {
        fprintf(stderr, "Failed to open output file\n");
        goto error;
    }
    if (fwrite(image_data, 1, pitch * res_height, output) != pitch * res_height) {
        fprintf(stderr, "Failed to write all data to output\n");
        goto error;
    }
    goto cleanup;
error:
    rc = 1;
cleanup:
    if (output) fclose(output);
    if (resource) vc_dispmanx_resource_delete(resource);
    if (image_data) free(image_data);
    if (display) vc_dispmanx_display_close(display);
    return rc;
}
```
This works in as much as it writes out the snapshot without any errors, but it isn't rotated. At least the *program* doesn't rotate it; if the system configuration specifies a rotation then the output is indeed rotated accordingly, although the dimensions of the resource must be adjusted accordingly (as the script demonstrates). At that point I thought "okay, rotation by system configuration is supported (or done by some layer before snapshot), but transforms on the snapshot itself aren't; they don't return an error when used, but don't do anything either, fair enough". Out of curiosity, though, I tried a 90 degree rotation:

```diff
--- snapshot_rotation.c 2016-11-14 10:04:53.395772805 +0000
+++ snapshot_rotation_90.c      2016-11-14 10:06:04.365679912 +0000
@@ -39,7 +39,7 @@
         goto error;
     }
     /* Try getting snapshot to rotate its output... */
-    if (vc_dispmanx_snapshot(display, resource, DISPMANX_ROTATE_180)) {
+    if (vc_dispmanx_snapshot(display, resource, DISPMANX_ROTATE_90)) {
         fprintf(stderr, "Failed to capture snapshot\n");
         goto error;
     }
```

This *doesn't* work (it fails to capture the snapshot with an error). Here I thought "hmm ... maybe it wants the size of the destination resource rotated?":

```diff
--- snapshot_rotation_90.c      2016-11-14 10:06:04.365679912 +0000
+++ snapshot_rotation_90_2.c        2016-11-14 10:07:56.985531328 +0000
@@ -19,13 +19,13 @@
     vc_dispmanx_display_get_info(display, &info);
     /* Transpose size if system configuration applies a 90/270 rotation */
     if (info.transform & 0x1) {
-        res_width = info.height;
-        res_height = info.width;
-    }
-    else {
         res_width = info.width;
         res_height = info.height;
     }
+    else {
+        res_width = info.height;
+        res_height = info.width;
+    }
     pitch = 3 * VCOS_ALIGN_UP(res_width, 16);
     image_data = malloc(pitch * res_height);
     if (!image_data) {
```

Nope, that produces an error when capturing the snapshot too. My assumption at this point is that there's *something* implemented for rotation but it's not exactly complete, so it's just documented as unsupported? If that is indeed the case, I wonder if I could be extremely cheeky and ask if someone could have a look to see if it would be a lot of work to support rotation in snapshots?

As a bit of background: my goal at the moment is to make a nice simple Python class in picamera for performing screenshots (ultimately this has to do with overlays on camera streams, but let's concentrate on the screenshot side for now). Currently, if the user's display isn't rotated this works nicely, and is extremely rapid. However, it's a slightly different story if the user's display *is* rotated. The snapshot does work, but the output of the snapshot is rotated (unsurprisingly)

I'm assuming this isn't what users will want (they've presumably set up a rotation on the display because their monitor actually is rotated, and I'm assuming what they'd *really* like is for the snapshot to come out oriented as they see things on their display). Thanks to the work on #348, I can now detect all rotations (and flips), but the question is how to compensate for them. I could [do what raspi2png does](https://github.com/AndrewFromMelbourne/raspi2png/blob/master/raspi2png.c#L367) and perform the reverse rotation myself in a couple of nested loops. That's not *too* bad in C, but in Python it's going to be horrifically slow (unless I rely on a third party lib like numpy which I'm rather hoping to avoid as a strict dependency). Hence my desire to get the GPU to do the heavy lifting again :)

Anyway - that's the background. If there's any other possibilities for resource rotation, I'm all ears! My only real desire is to avoid getting Python to do the rotations, so if there's something else in dispmanx (or mmal, or whatever) that I could use - great!

6by9, 2016-11-14:
One of the early sanity checks in the implementation of dispmanx_snapshot
```
  /* transform is now deprecated. */
  if (transform &~ (DISPMANX_SNAPSHOT_NO_YUV|DISPMANX_SNAPSHOT_NO_RGB|DISPMANX_SNAPSHOT_FILL) != 0 || image == 0)
  {
    vcos_assert (0);
    return DISPMANX_INVALID;
  }
```
DISPMANX_SNAPSHOT_NO_YUV drops all YUV layers from the snapshot. DISPMANX_SNAPSHOT_NO_RGB drops all RGB layers from the snapshot. 
DISPMANX_SNAPSHOT_FILL appears to stretch the snapshot to totally fill the supplied image buffer, but I wouldn't swear to that one (it's certainly munging dest_rect).

If any of the individual scene elements indicate a transform, then those should be applied in the snapshot.

JamesH65, 2017-12-05:
Any progress/fixes comes out of this or still ongoing?

JamesH65, 2018-07-02:
@waveform80 @6by9 Is this still ongoing? Not been much traffic!

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

JamesH65, 2018-10-29:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-10-29 by JamesH65

kiteretro, 2021-02-27:
Apologies for commenting on a closed issue, however this is something that I'm looking for.

I can also see that `vc_dispmanx_element_add` also has a transform arg and it too is unable to rotate 90 degrees in testing.

popcornmix, 2021-03-03:
The HVS cannot transpose images, but it can H and V flip them, so only those transforms are supported.