raspberrypi/userland #361: RaspiVid : --mode : sw error or my fault

bluerobert, 2016-12-14:
Hi,
I play with the original RPI cam and with the Model (H) from waveshare. For both I detect the same effect:
Dependent on the mode (1 or 5) the Angle of view differs:

Cam type     |        mode |  Angle of view
----------------|-----------------|----------------------
rpi                 |        5     |       93°
rpi                  |       1    |        73°
waveshare(H)  |     5   |         127°
waveshare(H)  |     1    |        90°

The angle was measured in the horizontal plane. Accuracy of the measurements rather +/- 8 °.
command : /usr/bin/raspivid -f -t 5000  --mode x  # x =  { 1,5 }
I also write the output to file. Same result.

Any comments ? Any ideas ?
Thanks.

6by9, 2016-12-14:
Please don't expect support here for the Waveshare camera module - contact them for any support.

Mode 1 is 1920x1080 pixels cropped out of the 2592x1944 of the sensor, so inherently has a narrower field of view.
The PiCamera library docs at http://picamera.readthedocs.io/en/release-1.12/fov.html#camera-modes are generally the clearest description of all the modes available. (The V2 8MPix camera module has more cropped modes than the V1 as 8MPix is actually a more awkward resolution to reduce to 1080P or 720P. 5MPix is actually a nice sweet spot)

bluerobert, 2016-12-14:
Hi 6by9,
Please read carefully my message. I only quoted the Waveshare cam to show that the behavior is the same with another cam also.

Many thanks for the documentation link and your description. I did not know that yet.

closed 2016-12-14 by bluerobert