raspberrypi/userland #362: result of mjpeg decoding using omxmjpegdec has many artifacts

sonntex, 2016-12-20:
I have usb camera that produces sequence of jpeg images that can not be decoded by omxmjpegdec correctly.

# ls -l
total 104
-rw-r--r-- 1 root root  7313 Jan  3 18:43 image-00.jpeg
-rw-r--r-- 1 root root  7339 Jan  3 18:43 image-01.jpeg
-rw-r--r-- 1 root root  7345 Jan  3 18:43 image-02.jpeg
-rw-r--r-- 1 root root  6835 Jan  3 18:43 image-03.jpeg
-rw-r--r-- 1 root root  6835 Jan  3 18:43 image-04.jpeg
-rw-r--r-- 1 root root  6835 Jan  3 18:43 image-05.jpeg
-rw-r--r-- 1 root root  6840 Jan  3 18:43 image-06.jpeg
-rw-r--r-- 1 root root 15832 Jan  3 18:43 image-07.jpeg
-rw-r--r-- 1 root root 15832 Jan  3 18:43 image-08.jpeg
-rw-r--r-- 1 root root 15862 Jan  3 18:43 image-09.jpeg

Pipeline with jpegdec (libjpeg or libjpeg-turbo backend) produces correct result that can be viewed by imagemagick.

# GST_DEBUG=6 gst-launch-1.0 -qe multifilesrc location="image-%02d.jpeg" caps="image/jpeg,framerate=30/1" ! jpegdec ! multifilesink location="image-%02d.yuv" 2>gst.log

# display -size 640x480 -depth 8 -sampling-factor 4:2:0 -colorspace srgb image-00.yuv

Pipeline with omxmjpegdec (rpi-userland backend) produces incorrect result that can be viewed by imagemagick and has many artifacts.

# GST_DEBUG=6 gst-launch-1.0 -e multifilesrc location="image-%02d.jpeg" caps="image/jpeg,framerate=30/1" ! omxmjpegdec ! multifilesink location="image-%02d-omx.yuv" 2>gst-omx.log

# display -size 640x480 -depth 8 -sampling-factor 4:2:0 -colorspace srgb image-00-omx.yuv

Is it problem of rpi-userland or gst-omx?

I filed same bug to gstreamer bug tracker. You can find it at https://bugzilla.gnome.org/show_bug.cgi?id=776311

Attachment has image-00.jpeg, image-00.yuv and image-00-omx.yuv files from my example above.

[images.zip](https://github.com/raspberrypi/userland/files/663978/images.zip)

.

JamesH65, 2016-12-20:
Are those supposed to be all black results?

sonntex, 2016-12-20:
Yes! All images were made in a room without any lighting.

6by9, 2016-12-20:
Please confirm firmware, kernel, and GStreamer versions.
```vcgencmd version```
```uname -a```
```gst-launch-1.0 --version```
The top section of ```gst-inspect-1.0 omxmjpegdec``` would be good too (I'm expecting that to be the right component name, but could be wrong).

sonntex, 2016-12-20:
```
# vcgencmd version
May 20 2016 19:01:31 
Copyright (c) 2012 Broadcom
version faf071dd4885c5ac1a89483d35a5326e7f69495f (clean) (release)
```

```
# uname -a
Linux buildroot 4.4.15-v7+ #4 SMP Wed Sep 21 16:58:47 MSK 2016 armv7l GNU/Linux
```

```
# gst-launch-1.0 --version
gst-launch-1.0 version 1.8.2
GStreamer 1.8.2
Unknown package origin
```

```
# gst-inspect-1.0 omxmjpegdec
Factory Details:
  Rank                     primary + 1 (257)
  Long-name                OpenMAX MJPEG Video Decoder
  Klass                    Codec/Decoder/Video
  Description              Decode MJPEG video streams
  Author                   Sebastian Dröge <sebastian.droege@collabora.co.uk>

Plugin Details:
  Name                     omx
  Description              GStreamer OpenMAX Plug-ins
  Filename                 /usr/lib/gstreamer-1.0/libgstomx.so
  Version                  1.2.0
  License                  LGPL
  Source module            gst-omx
  Source release date      2014-07-23
  Binary package           GStreamer OpenMAX Plug-ins source release
  Origin URL               Unknown package origin

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstVideoDecoder
                         +----GstOMXVideoDec
                               +----GstOMXMJPEGDec
                                     +----GstOMXMJPEGDec-omxmjpegdec

Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      image/jpeg
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]

  SRC template: 'src'
    Availability: Always
    Capabilities:
      video/x-raw
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]
              framerate: [ 0/1, 2147483647/1 ]

Element Flags:
  no flags set

Element Implementation:
  Has change_state() function: gst_omx_video_dec_change_state

Element has no clocking capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  name                : The name of the object
                        flags: readable, writable
                        String. Default: "omxmjpegdec-omxmjpegdec0"
  parent              : The parent of the object
                        flags: readable, writable
                        Object of type "GstObject"
```

sonntex, 2016-12-20:
Also I have gstreamer logs from my example above.

[logs.zip](https://github.com/raspberrypi/userland/files/664290/logs.zip)

sonntex, 2016-12-28:
Do you have any news?

6by9, 2017-01-04:
Sorry, not really had a chance to dig into this as yet.
The corrupt lines look like a cache flushing/invalidation issue which could be either side. The noisy macroblocks on the first two lines are likely to be something else.

6by9, 2017-07-12:
Only just getting around to investigating this - sorry for the delay.

Where were you getting your version of omxmjpegdec from? I've just installed it on Raspbian using apt-get and get 
```
pi@raspberrypi:~/userland-362 $ gst-inspect-1.0 omxmjpegdec
Factory Details:
  Rank                     primary + 1 (257)
  Long-name                OpenMAX MJPEG Video Decoder
  Klass                    Codec/Decoder/Video
  Description              Decode MJPEG video streams
  Author                   Sebastian Dröge <sebastian.droege@collabora.co.uk>

Plugin Details:
  Name                     omx
  Description              GStreamer OpenMAX Plug-ins
  Filename                 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstomx.so
  Version                  1.0.0.1
  License                  LGPL
  Source module            gst-omx
  Source release date      2015-11-11 17:35 (UTC)
  Binary package           GStreamer OpenMAX Plug-ins git
  Origin URL               Unknown package origin
```
I've created a set of JPEGs using Raspistill in a similar manner to you. Running
```
GST_DEBUG=6 gst-launch-1.0 -qe multifilesrc location="output%d.jpg" caps="image/jpeg,framerate=30/1" ! omxmjpegdec ! multifilesink location="output-omx-%02d.yuv"
```
I get 3 files, and the log has stopped on:
```
0:00:04.029734504   628 0x72802400 DEBUG            omxvideodec gstomxvideodec.c:2801:gst_omx_video_dec_loop:<omxmjpegdec-omxmjpegdec0> Read frame from component
0:00:04.029825962   628 0x72802400 DEBUG            omxvideodec gstomxvideodec.c:2803:gst_omx_video_dec_loop:<omxmjpegdec-omxmjpegdec0> Finished frame: ok
0:00:04.029891014   628 0x72802400 DEBUG                    omx gstomx.c:1257:gst_omx_port_acquire_buffer:<omxmjpegdec-omxmjpegdec0> Acquiring resize buffer from port 61
0:00:04.029961691   628 0x72802400 DEBUG                    omx gstomx.c:1357:gst_omx_port_acquire_buffer:<omxmjpegdec-omxmjpegdec0> Queue of resize port 61 is empty
```
Having just had a rummage around with gstomx, it does stick a resize after video_decode mainly to allow format conversion. 

Each of my files are also 1228800bytes despite only being VGA. They're RGBA rather than YUV. However they are all valid.

Taking your jpeg, copying a couple of times to give me a stream, and then running with that gives me a set of images which aren't black. They're corrupted versions of my previous set of images taken with raspistill. So it looks like something is using uninitialised memory.

Running your JPEG through JPEGSnoop (http://www.impulseadventure.com/photo/jpeg-snoop.html) it confirms that your JPEG is corrupt. Checking with the GPU debugger, the codec is throwing errors so either the firmware isn't propogating them (which is correct behaviour under some circumstances as it isn't fatal), or the application isn't listening for them.
Summary though appears to be that your files aren't valid. Could you post a few more so that we can confirm that?

JamesH65, 2017-12-05:
@sonntex We'll need some feedback on this, otherwise I will close the issue.

sonntex, 2017-12-05:
I switched to nanopi and did not work with rpi for a long time. I do not have any new information for this issue. You should close the issue.

closed 2017-12-05 by JamesH65

jetibest, 2021-11-07:
I got this issue too. Some white/black/green horizontal bars with patterns and traces of frame mixing occur in my pipeline. It took me a while to realize it was omxmjpegdec that was the culprit, swapping `omxmjpegdec` to `jpegdec` fixed the issue for me.

My pipeline is as follows:

    gst-launch-1.0 -q v4l2src device=/dev/video0 ! image/jpeg,width=1280,height=720 ! omxmjpegdec ! video/x-raw,format=I420 fdsink

I noticed this issue occurring on RPi 3B, but not on RPi 3B+. Later, I discovered, that increasing CPU usage results in these artifacts, can reproduce it consistently on the RPi 3B. So I ran three instances of infinite while-loops in bash, and the gstreamer instance, and this would lead to constant flickering in the videostream due to these artifacts.

I am using the latest version of all of the given software, using Arch Linux.

It definitely has to do with same memory being overwritten, while it still needs to be flushed, or a similar thing. If you could point me to the relevant source code, I can perhaps look into this myself as well.