raspberrypi/userland #363: Preview Full HD Pi Camera don't show the hole picture

fabiocolbrelli, 2017-01-09:
Hello Raspberry Team

I have an issue with the preview of the pi camera. Here is the code im using:

camera = PiCamera()
camera.resolution = (1920, 1080)
camera.start_preview()
camera.capture(os.path.join(location,filename))

The resolution of my screen is 1920x1080, when I start the preview i see a live picture of the camera, when the picture gets taken, on the picture I can see a lot more then on the preview. So the preview is not using the full picture, the image seems to be zoomed in.

When I do the same with a smaller resolution:
camera = PiCamera()
camera.resolution = (1280, 720)
camera.start_preview()
camera.capture(os.path.join(location,filename))

The preview starts fullscreen even if the resolution on the screen is still 1920x1080, and finally I see the hole picture on the screen. The down side about this fix is, that even the image has only the resolution of 1280x720 and it would be nice to use the full resolution of the camera.

A nice way would be to could set up a preview resolution and taking resolution, the preview should be like HDready and the picture FullHD something like that ;) So it would be possible even with a screen with a bader resolution to take a picture with the full resolution.

A workarround I used is this:
camera = PiCamera()
camera.resolution = (1280, 720)
camera.start_preview()
when taking the picture:
camera.resolution = (1920, 1080)
camera.capture(os.path.join(location,filename))
camera.resolution =(1280, 720)

The problem with this it closes the preview take a picture and opens again the preview.

Friendly greetings and thank you for your help
fc_key

fabiocolbrelli, 2017-01-09:
https://www.raspberrypi.org/forums/viewtopic.php?uid=185969&f=28&t=63722&start=0

6by9, 2017-01-09:
http://picamera.readthedocs.io/en/release-1.12/fov.html#camera-modes

The 1920x1080 mode off the sensor has a reduced field of view as the sensor can't read the full frame fast enough and with enough pixels to reach 1080P30.
I seem to recall @waveform80 added a way to run preview at an alternate resolution to the capture in a recent release of PiCamera, but I can't recall the syntax or see the relevant docs with a quick browse.

fabiocolbrelli, 2017-01-09:
Thank you for the quick answer...It would be nice to see the 1080preview even with only 15fps...is there a posibility?

6by9, 2017-01-09:
Ask for 15fps (```framerate```parameter) and force mode 2 (```sensor_mode``` parameter).

waveform80, 2017-01-09:
Please note that in the current picamera release the `sensor_mode` parameter has an issue (setting it after construction doesn't work). It's already been corrected and the fix will be in the next release (hopefully relatively shortly).

fabiocolbrelli, 2017-01-09:
Okay I will try it when I'm back at home :)

fabiocolbrelli, 2017-01-09:
I tried it the different chart that i found here helped me to understand how the camera and the preview works, but I dont get why on Full HD screen you cant see the Full HD preview but you can see loads of bigger resolution with a smaller preview for example 3280x2464 without any problem.

What I tried is this:
camera.resolution = (1920, 1080)
camera.framerate = 15
camera.sensor_mode = 5
camera.start_preview()

The preview is still zoomed in it seems exactly the same as before, is this because the sensor_mode command dont work as expected.

Thank you for your time :)

Friendly greetings fc_key

fabiocolbrelli, 2017-01-09:
http://picamera.readthedocs.io/en/release-1.12/fov.html?highlight=sensor%20mode Here I found the informations to the resolutions and preview

fabiocolbrelli, 2017-01-11:
@waveform80 Can you please explain me a second how to use sensor_mode right? Or doesn't the function work at all right now till the next release?

So you can set the camer.resolution to a bigger resolution and show the preview in a smaller resolution?

Thank you a lot

Friendly greetings fc_key

6by9, 2017-01-12:
```
camera = PiCamera(resolution=(1920, 1080), framerate=15,sensor_mode=5)
camera.start_preview()
```
That is providing the information to the constructor. You were trying to change it afterwards, which is what waveform80 had reported as having the issue.

fabiocolbrelli, 2017-10-25:
Hello to all of you can you update me about the preview status of the pi camera was there something changed? what is the highest resolution i can do a full preview on the screen with a framerate of 30fps?

Thank you friendly greetings
fc_key

6by9, 2017-10-26:
Nothing has changed or will change if you want full field of view.

As previously linked, http://picamera.readthedocs.io/en/latest/fov.html#sensor-modes lists out all the sensor modes. Those are typically limited by the hardware, so nothing will change there.
If you want full field of view and 30fps, then you have to use a binned mode. On a V1 sensor that means a max of 1296x972 off the sensor, or on a V2 sensor a max of 1640x1232. (The SoC will resize this any way you want to, so it doesn't have to be exactly those resolutions).

Within PiCamera it is possible to set the preview resolution independently of the capture resolution. http://picamera.readthedocs.io/en/latest/api_renderers.html#pipreviewrenderer appears to be the relevant piece of the docs (I can't see if there is an easier mechanism for setting preview resolution)

JamesH65, 2017-12-05:
This is a wont fix since there is nothing to fix. Please use the forum for questions of this type - not actually a bug, just a usage issue.

closed 2017-12-05 by JamesH65