raspberrypi/userland #365: Raspiyuv: When using luma and time lapse, the size is double after the first frame

dane-martel, 2017-01-13:
When using raspiyuv with the luma option and time lapse, the frame sizes are double what they should be after the first frame.

command:
`raspiyuv -y -v -o frame2%08d.nv21 --width 640 --height 480 -ss 1000000 -n -ISO 100 -set -tl 0 -t 0 -ex off -awb off -ifx none -drc of`

file sizes:
-rw-r--r--  1 pi pi 307200 Jan 13 17:22 frame200000001.nv21
-rw-r--r--  1 pi pi 614400 Jan 13 17:22 frame200000002.nv21
-rw-r--r--  1 pi pi 614400 Jan 13 17:22 frame200000003.nv21
-rw-r--r--  1 pi pi 614400 Jan 13 17:22 frame200000004.nv21

6by9, 2017-01-14:
Thanks for the report.

I haven't got a Pi to hand to test, but I think the issue is going to be that https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStillYUV.c#L582 should read
```bytes_to_write = vcos_min(port->format->es->video.width * port->format->es->video.height, buffer->length);```
The camera component sends the image data in one buffer and ```MMAL_BUFFER_HEADER_FLAG_FRAME_END``` flag, and then a buffer with no data but the ```MMAL_BUFFER_HEADER_FLAG_EOS``` flag set. The current code will be writing width*height bytes of undefined data due to that.
If you can then please give that a try, otherwise I'll try to find a moment in the next few days.

NB The frames aren't NV21. That is a full YUV 4:2:0 frame with all the Y values, followed by pixel interleaved V and U values. Y800, Y8, GREY are the normal 4CC for 8bpp luma only / greyscale (http://fourcc.org/yuv.php and http://fourcc.org/pixel-format/yuv-y800/)

JamesH65, 2017-01-14:
If I have the chance, I'll take a look at this sometime next week.

6by9, 2017-01-14:
@JamesH65 I thought you might, but didn't want to presume :-)

JamesH65, 2017-01-16:
Been looking a bit at this. There is an inherent issue with the modification to save only luma in that the code assumes that the luma is contained entirely in one buffer. If it is split over multiple buffers, it does not keep count of how much has already been saved from previous buffers. That's in addition to the issue defined here, which 6x9's change above should fix. It's quite possible that the Luma will never be spread over multiple buffers, in which case the fix is OK for all cases.

6by9, 2017-01-16:
```MMAL_ENCODING_I420``` inherently is a full frame.
There is ```MMAL_ENCODING_I420_SLICE``` (and relatives) which define a 16 line high slice, but they are only supported on a few components, and are not what we're dealing with here.

So it sounds like I was right with my prediction ;-)

JamesH65, 2017-01-17:
Cool, I'll make the change and give it a try. If I can get hold of a camera.

ethanol100, 2017-01-17:
Yes, it is the correct prediction and 6by9's change will fix the issue. There is always one buffer with flag 4 and the image data and then one with flag 1 and no image data. Additionally setting -tl 0 switches to  FRAME_NEXT_IMMEDIATELY which will wait 1s for the first wait and then 30ms for all the following. With a long enough waiting time between the two buffers, the file handle is still closed, and the second buffer of length 0 is not written. This explains why the first image had the right size and all following the wrong one. And it explains why i.e. -tl 1000 did create the correct files, too.

6by9, 2017-01-17:
It may be your explanation, but I'm not sure I quite agree with your reasoning there @ethanol100.

Buffer 1. Len = size, Flags = MMAL_BUFFER_HEADER_FLAG_FRAME_END
  Saves ```width*height``` bytes to file 1. https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStillYUV.c#L601 is true, so sets the semaphore for file to be changed.
File 1 closed and file 2 opened.
Buffer 2. Len = 0, Flags = MMAL_BUFFER_HEADER_FLAG_EOS
  Saves ```width*height``` bytes of invalid data (1) to file 2.
Buffer 3. Len = size, flags = MMAL_BUFFER_HEADER_FLAG_FRAME_END
  Saves ```width*height``` bytes to file 2. Semaphore set for file to be changed.
File 2 closed and file 3 opened.
Buffer 4. Len = 0, Flags = MMAL_BUFFER_HEADER_FLAG_EOS
  Saves ```width*height``` bytes of invalid data to file 3.
etc.

```-tl 1000``` produced the correct files? I hadn't seen that reported, but if that did produce the correct files I can only see it as being from a race condition of whether the file is closed and reopened before or after the EOS buffer comes back. If there is a file handle open when the EOS buffer comes back, it would write to it.

(1) https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStillYUV.c#L836 ```still_port->buffer_num = still_port->buffer_num_recommended;```, which my memory says is 1. So the data will actually be the same as for the previous frame. Still the wrong thing to do, but will look like valid data.

ethanol100, 2017-01-17:
The file handle is set to "NULL" directly after pData->complete_semaphore is posted and will only be reopen after the next wait_for_next_frame has finished in the main loop. The data is only written, if the file handle is still valid in the buffer callback. It is a race condition, between opening the new file and receiving the buffer.

Anyway, your supposed fix corrects the issue. I just tried to argue, why the first frame had the correct size.

6by9, 2017-01-17:
Quite correct, I'd missed the handle getting set to NULL. Always good to have extra sets of eyes on these things.

6by9, 2017-02-08:
Are we happy that https://github.com/raspberrypi/userland/pull/366 resolves this? in which case this can be closed.

closed 2017-02-08 by dane-martel