raspberrypi/userland #371: Raspivid changing bitrate on fly?

Triuman, 2017-01-26:
Hi,

I'm streaming H264 from rpi camera using raspivid and Gstreamer and I need to change the bitrate dynamically to adapt to network conditions. Is it possible for you to implement this feature?

Thank you.

6by9, 2017-01-26:
Assuming you're just piping the output of raspivid into gstreamer, then no.

Use V4L2 as it should already support it, and GStreamer has a v4l2src component.
You may need to use a recent version of GStreamer as they didn't like our H264 stream via V4L2 (see https://github.com/raspberrypi/linux/issues/849), but GST 1.9 is reported as all being good.

Triuman, 2017-01-26:
I'm actually using [gst-rpicamsrc](https://github.com/thaytan/gst-rpicamsrc) which is a GStreamer wrapper around the raspivid/raspistill. He [here](https://github.com/thaytan/gst-rpicamsrc/issues/60#issuecomment-275301949) says firmware should support it first.

I can't open raspberry pi camera using v4l2src, only usb cameras. And it apparently doesn't have a bitrate property to set. I'll look into it deeper though. Thank you.

JamesH65, 2017-01-26:
You need to install the specific V4L2 module for the RPF camera, then it
should work. haven't got link to hand - google will find it.

6by9, 2017-01-26:
The firmware has always supported it using MMAL_PARAMETER_VIDEO_BIT_RATE - that's the parameter the V4L2 driver uses.

```sudo modprobe bcm2835-camera``` will load the V4L2 driver for the Pi camera. It's not loaded by default as it adds a resource overhead on the GPU side whenever it is loaded which many won't want.
It supports V4L2_CID_MPEG_VIDEO_BITRATE for bitrate if using H264 or MJPEG encoding.

Triuman, 2017-01-26:
`sudo modprobe bcm2835-camera` gives the error "modprobe: FATAL: Module bcm2835-camera not found." . I tried `sudo modprobe bcm2835-v4l2` and `gst-launch-1.0 -v v4l2src` didn't give any error but it apparently get no stream.

the pipeline I try is;

` gst-launch-1.0 -v v4l2src ! 'video/x-h264,width=640,height=480,framerate=40/1' ! queue ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=127.0.0.1 port=8004`

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ famerate\=\(fraction\)40/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/\,\ framerate\=\(fraction\)40/1"
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ frameate\=\(fraction\)40/1"
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1,\ framerate\=\(fraction\)40/1"
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ framrate\=\(fraction\)40/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)11\,\ framerate\=\(fraction\)40/1"
```

I tried also putting  "bcm2835-v4l2" into /etc/modules but it didn't work either and actually broke my rpicamsrc pipeline.

Is there anything else I should do?

(Thanks for your support. If it's not the right place to discuss this, I can move to rPI forums.)

6by9, 2017-01-26:
Apologies for the typo (now corrected in my original comment).
It sounds like you're using an older version of GStreamer, hence hitting the same issue I'd referenced before. For some reason it seems to fail to acknowledge that there are valid header bytes in the stream.

Seeing your latest comment on https://github.com/thaytan/gst-rpicamsrc/issues/60 I'm going to close this as you have a solution with gst-rpicamsrc supporting dynamic bitrate changing. There's no need for any changes to userland. For extra support on V4L2 then the forums are probably the best place.

closed 2017-01-26 by 6by9