raspberrypi/userland #375: [RFC]hello_pi examples: make them work with kms/fkms/vchiq

Electron752, 2017-02-12:
I've been thinking about the hello_pi examples.  I was wondering if it would be possible to modify the userland libraries in opt/vc to be a thin wrapper on top of the mesa based solution instead of vchiq.   The idea would be that if kms or fkms is loaded, the calls to the legacy brcm usermode APIs get routed to mesa based libraries.  If kms or fkms is not loaded, then the calls get routed to the old vchiq based solution.  The idea being that the vchiq based solution goes away after awhile.

For the most part, I think this makes sense.  It would of course need a method to tell if kms or fkms is loaded instead.  Otherwise it's just a bunch of userland glue.

I would be interested in helping with this, but I don't know that much about the 3D graphics userland libraries.  But I would be willing to learn.

@anholt and @popcornmix : Any thoughts on this approach?

anholt, 2017-02-13:
I had talked to Eben about the idea, and his response was "why do we need them to keep working, at that point?"  If we have a normal GL implementation, then all the other demos in the world now work and Raspberry Pi doesn't need their own.

Electron752, 2017-02-14:
That makes sense...

The only thing I could find that's missing is minecraft_pi.  It requires vchiq to work and it appears that it's using the old broadcom libraries to do graphics.   So minecraft_pi would need to be ported or at least the rpi specific sections would need to be removed.

JamesH65, 2017-12-05:
Closing as a wont fix. Minecraft is supplied by MS, so they will need to fix it up as necessary.

closed 2017-12-05 by JamesH65