raspberrypi/userland #377: Use nproc instead of 6 in buildme

LukeSkyw, 2017-03-01:
Some people have more than 6 cores on their CPU.
Why not use nproc in buildme like so:
make -j `nproc`

6by9, 2017-03-01:
Agreed that that would be a sensible thing to do.
Feel free to create a pull request that will change it.

pelwell, 2017-03-01:
Isn't it the case that nproc*1.5 is a better heuristic?

6by9, 2017-03-01:
TBH userland is generally fairly quick to compile anyway. Using nproc would avoid crippling a machine with <4 cores.
How i/o bound is the "generic system"? Do you assume the data is on an SSD and therefore very swift, or is it a slow HDD in which case a larger multiple of nproc would be appropriate.

I think everybody has their own rule - nproc+1 is another I know to be quite popular. I think this is a case of choose something sensible.

JamesH65, 2017-03-02:
Once upon a time I did actually do some testing, and x1.5 was spot on, but
that was spinning rust storage, SSD's may well be different. Note the
difference was x1 to x1.5 was fairly small but there.

Ferroin, 2017-03-02:
There are too many variables involved.  For traditional HDD's on IDE or SATA, the ideal number usually is about 1.5 when using GCC, but it goes down when using clang (Clang compiles things faster, so it spends more time waiting on IO, but having more processes waiting on IO slows things down).  On most SCSI HDD's, it's actually almost 2 for GCC.  On SATA and SCSI SSD's it's typically 1+nproc, and for NVMe SSD's or NVDIMM's (both of which are relatively rare still), it's actually exactly nproc.  IOW, the faster your storage and the faster your compiler, the lower the ideal value.

The standard for most projects that automatically parallelize though is exactly nproc, since it means they can take advantage of the whole CPU, but aren't likely to edge out other stuff running on the system.

commit ea29d9ac (2017-04-05): Use nproc build jobs

Fixes #377.
Replace hard coded -j6 in make lines for -j `nproc`
to adopt the same number of jobs as processor cores.

Discussion abounds as to whether there is a better
multiple for CPU vs I/O speeds, but it's better than
always using 6.
closed 2017-04-05 by popcornmix