raspberrypi/userland #385: feature suggestion for raspivid: timestamp in file name option when recording in segments

jdesbonnet, 2017-03-23:
I am using raspivid to record in segments, but I need to preserve the exact time the recording started (file lastmodified stamps are easily changed by accident). A nice option would be to use a timestamp in the filename instead of a sequential counter.

jdesbonnet, 2017-04-11:
I've got a very small modification that achieves this:
https://github.com/jdesbonnet/userland/tree/segmented_recording_filename_timestamps

JamesH65, 2017-04-11:
> recording_filename_timestamps

Interesting idea that has no impact elsewhere. Thoughts @6x9 ?

Would need a small code tidy and the docs updating...!!

6by9, 2017-04-11:
Yes, other than a couple of formatting issues that I've just commented on it looks fine. Please create a pull request for it (I haven't checked that your branch is based off the top of tree, but the change cherry-picked fine)

I can't quite believe that you can't printf a time_t without a cast, but it appears that way. The cast on tv_nsec can be avoided with %03ld, but in this case it makes little difference due to the divide.

jdesbonnet, 2017-04-11:
Thanks for the feedback. Sorry, that was just interned as an initial hack to see if was an easy thing to do. Ideally the user should be able to pass the date formatting string in standard C library format, but the simple test for "%d" vs "%s" won't work in that case. Perhaps that over complicates it a bit.

JamesH65, 2017-12-05:
@jdesbonnet Are you planning a PR to implement this?

JamesH65, 2018-07-02:
@jdesbonnet We would need a PR to make this change.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

JamesH65, 2018-07-20:
Fixed by #479

closed 2018-07-20 by JamesH65