raspberrypi/userland #390: Can we fix the handling of the different variants of MMAL_ENCODING_OPAQUE?

6by9, 2017-04-20:
Under the bonnet there are 3 forms of opaque buffer. Most forms are accepted by the majority of components, but the camera->output[2] (stills port) produces a form that is not widely accepted.
At the moment if connected to (eg) null_sink then the pipeline stalls.

Under some circumstances (all but the low memory modes) it should be possible to avoid stalling the pipeline if the MMAL framework does some juggling of the buffers.

6by9, 2017-07-12:
Firmware fix sorted. Will be released in next rpi-update.

6by9, 2017-10-30:
Reopening as #435 has flagged up the single full frame to stripe wasn't handled correctly.
Image_encode also has issues if presented YUVUV full frames, so we possibly need a more complete review of all combinations.

JamesH65, 2018-07-02:
@6by9 The linked issue has been closed as fixed - does that mean this is also fixed?