raspberrypi/userland #391: Add bcm_host library support for arm64

raidensnake, 2017-04-21:
I ran the recent userland on debian arm64 and it was missing the required bcm_host library needed to compile retropie's rpi optimised SDL libraries. I was able to compile the library manually in the source folder with no issues. Maybe it would help by adding that to the cmake lists so it can be compiled in arm64.

6by9, 2017-04-21:
Create a pull request with your proposed change and it will be reviewed.

We're not prioritising arm64 development but are happy to work with others who wish to try to get it working. There are various places where interfacing with the GPU firmware may be "fun" as that is only 32bit, so any pointers in shared structs are going to mismatch in size.

raidensnake, 2017-04-22:
what I'm saying is the bcm_host library compiles as is on arm64. but it has to be done manually.

stiltr, 2017-07-25:
I needed bcm_host for libcec on ARM64 and was able to compile it by removing the exclusion from host_applications/linux/CMakeLists.txt. Everything _seems_ to work ok from what I can tell (libcec works fine). Is there a reason this is being excluded on ARM64? I'm happy to create a PR if needed, but "it worked for me" isn't exactly "fully tested".

JamesH65, 2017-12-05:
@stilr A PR enables us to look at the code quickly to see what will change and check for any obvious issues, so that would be a good start.

raidensnake, 2018-02-04:
Since I don't really know any of that stuff all I did was go into the host_applications/linux/libs/bcm_host manually and use cmake to compile it.

luked99, 2018-02-04:
On 4 February 2018 at 16:42, raidensnake <notifications@github.com> wrote:
>
> Since I don't really know any of that stuff all I did was go into the host_applications/linux/libs/bcm_host manually and use cmake to compile it.

Isn't the problem that VideoCore exposes a 32 bit API, where all the
handles are 32 bit. So although you might be able to just recompile,
it won't actually work when you pass a 64 bit pointer across and
expect it to come back in one piece.

I think various schemes have been suggested for fixing this, but
no-one has invested the time to turn it into a pull request.

Or perhaps it's more complicated than that.

6by9, 2018-02-05:
> Isn't the problem that VideoCore exposes a 32 bit API, where all the handles are 32 bit. So although you might be able to just recompile, it won't actually work when you pass a 64 bit pointer across and expect it to come back in one piece.
I think various schemes have been suggested for fixing this, but no-one has invested the time to turn it into a pull request.
Or perhaps it's more complicated than that.

That pretty much sums it up.

The VCSM changes in that branch look pretty broken as they're changing userspace to kernel ABI, and will be changing the required behaviour of all clients using the library in some cases
```
-unsigned int vcsm_vc_hdl_from_hdl( unsigned int handle );
+unsigned long vcsm_vc_hdl_from_hdl( unsigned long handle );
```
VC handles are 32 bit. No reason to increase that to 64 other than to cause confusion.

I suspect that someone has seen the bit in MMAL that assigns the VC handle to the data pointer (which is now 64 bit) and worked backwards that the VC handle needs to be 64 bit, when in reality it needs a cast (and lots of magic to map between the two sizes of ```MMAL_BUFFER_HEADER_T``` due to the size of the pointer changing).

bcm_host may build and work correctly, but really don't expect MMAL or VCSM to work based on this.

raidensnake, 2018-03-29:
sure.

JamesH65, 2018-07-02:
Closing this issue as questions answered/issue resolved.

closed 2018-07-02 by JamesH65

alien999999999, 2023-03-03:
> I needed bcm_host for libcec on ARM64 and was able to compile it by removing the exclusion from host_applications/linux/CMakeLists.txt. Everything _seems_ to work ok from what I can tell (libcec [...]

Sorry to hijack, but i'm trying libcec on aarch64 mainline also, and am missing bcm_host, but also vchiq_arm ; do you have a pointer how you got libcec to work, bcm_host for this i get, but how to get the vchiq_arm library too?