raspberrypi/userland #393: CPU detection.

ED6E0F17, 2017-04-24:
hello_fft relies on "/proc/device-tree/axi"
(from https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c#L164)
which no longer exists.

pelwell, 2017-04-24:
Yes, annoyingly that is the one piece of information that was used in the "axi" subtree, and it is dynamic in the sense that its value depends on the `disable_l2cache` config setting. It's probably necessary to restore this one property.

popcornmix, 2017-04-26:
Can you test latest rpi-udpate firmware?

ED6E0F17, 2017-04-26:
Working as expected. After updating just the firmware: /proc/device-tree/axi is present, and my Pi2 has a working hello_fft with "#define GPU_FFT_USE_VC4_L2_CACHE 1" enabled (set for Pi1 use).

closed 2017-04-26 by ED6E0F17