raspberrypi/userland #394: Supported libraries on arm64

carlonluca, 2017-05-03:
Is there an "official" up-to-date document describing what libraries in this repo are supported on arm64? Like EGL, OpenMAX, OpenGL etc...? I see something was ported and something else is not but I'm not able to find any document summarising the status.

popcornmix, 2017-05-04:
Officially arm64 is not supported. It is not generally tested by us and no promises are made.
We do sometimes accept patches to the kernel or userland libraries that improve arm64 support as long as there is no harm to arm32. But that is just as a convenience for early testers.

I think currently expect no gpu acceleration to work. In general EGL, OpenMAX, OpenGL etc do tend to pass 32-bit user pointers in and get them returned through callbacks that causes an API issue for arm64.

You may find the arm side GL driver does work with arm64 (I've heard suggestion it does without testing it myself) and as that is the way forward we're unlikely to put much effort into GPU side GL support.

We are looking to get camera and video decode working alongside the arm GL driver. Possibly through mmal, possibly through V4L2 and that may end up being a scheme that is supported in the arm64 world.

carlonluca, 2017-05-06:
So almost no acceleration is currently available on arm64. It is a pity that you do not provide arm64 version strictly of the closed source portion allowing the community to contribute to the rest, but I understand you probably consider the cost too high. Having OpenGL, OpenMAX and EGL on arm64 would be great. Thank you for the clarification anyway!

closed 2017-05-06 by carlonluca