raspberrypi/userland #396: missing include header on interface/mmal/mmal_buffer.h

zchee, 2017-05-17:
Hi,

Now, I writing libmmal bindings for Go language. Because I want to(writing) manage the raspberrypi's camera in Go. similar python-picamera.
But I have issue in `interface/mmal/mmal_buffer.h` header file.

I'll describe it just in case, Go language can include and use C header file. (called **cgo** in Go)
e.g.: https://golang.org/misc/cgo/stdio/stdio.go.
That's included `#include <stdio.h>` and called `C.getStdout()` function (actually C's `getStdout()`). similar Python `ctypes.CDLL` but use header file.

---

OK, back to the main topic. I wrote below code.

```go
package mmal

/*
#include <interface/mmal/mmal_buffer.h>
*/
import "C"

type BufferHeaderVideoSpecific struct {
	c C.MMAL_BUFFER_HEADER_VIDEO_SPECIFIC_T
}
```

That's can't compile. The compile error message(from gcc) are:

```console
In file included from go/src/github.com/zchee/go-picamera/mmal/buffer.go:8:0:
/opt/vc/include/interface/mmal/mmal_buffer.h:206:22: error: expected declaration specifiers or '...' before '*' token
 typedef MMAL_BOOL_T (*MMAL_BH_PRE_RELEASE_CB_T)(MMAL_BUFFER_HEADER_T *header, void *userdata);
                      ^
/opt/vc/include/interface/mmal/mmal_buffer.h:216:74: error: unknown type name 'MMAL_BH_PRE_RELEASE_CB_T'
 void mmal_buffer_header_pre_release_cb_set(MMAL_BUFFER_HEADER_T *header, MMAL_BH_PRE_RELEASE_CB_T cb, void *userdata);
                                                                          ^
/opt/vc/include/interface/mmal/mmal_buffer.h:227:1: error: unknown type name 'MMAL_STATUS_T'
 MMAL_STATUS_T mmal_buffer_header_replicate(MMAL_BUFFER_HEADER_T *dest, MMAL_BUFFER_HEADER_T *src);
 ^
/opt/vc/include/interface/mmal/mmal_buffer.h:235:1: error: unknown type name 'MMAL_STATUS_T'
 MMAL_STATUS_T mmal_buffer_header_mem_lock(MMAL_BUFFER_HEADER_T *header);
 ^
```

But if changed `#include <interface/mmal/mmal.h>` in Go code, can compile.
I think `mmal_buffer.h` have not included any local header file such as`mmal_types.h` etc (not yet tested).
Is it not supported to be used alone? or Am I wrong?

If this issue on the `raspberrypi/userland` side, I will send a PR.
Thanks.

pelwell, 2017-05-17:
Although I could not find it documented, the expectation is that applications always include `mmal.h` before any extra MMAL headers that might be needed. For example, [mmal_buffer.c](https://github.com/raspberrypi/userland/blob/master/interface/mmal/core/mmal_buffer.c#L28) begins:
```
#include "mmal.h"
#include "mmal_buffer.h"
#include "core/mmal_buffer_private.h"
#include "mmal_logging.h"
```
My belief is that this was considered to be cleaner than cross-including from all of the individual headers.

zchee, 2017-05-21:
@pelwell Sorry for the late reply. and Thanks.

> My belief is that this was considered to be cleaner than cross-including from all of the individual headers.

I see. So, might be needs below code(but `mmal.h` already included `mmal_buffer.h`...?)

```go
package mmal

/*
#include <interface/mmal/mmal.h>
#include <interface/mmal/mmal_buffer.h>
*/
import "C"

type BufferHeaderVideoSpecific struct {
	c C.MMAL_BUFFER_HEADER_VIDEO_SPECIFIC_T
}
```

Anyway, I understand why always include `mmal.h` header.
Thanks. Close.

---

BTW, I writing [zchee/go-mmal](https://github.com/zchee/go-mmal).
Please check this if you have an interest for Go.

closed 2017-05-21 by zchee