raspberrypi/userland #398: vchiq_lib_init: Conditional jump or move depends on uninitialised value(s)

minfrin, 2017-05-23:
Valgrind is reporting this:

    ==14464== Conditional jump or move depends on uninitialised value(s)
    ==14464==    at 0x6C27D98: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
    ==14464==  Uninitialised value was created by a stack allocation
    ==14464==    at 0x6C27CDC: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
    ==14464== 
    ==14464== Conditional jump or move depends on uninitialised value(s)
    ==14464==    at 0x6C27DA4: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
    ==14464==  Uninitialised value was created by a stack allocation
    ==14464==    at 0x6C27CDC: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
    ==14464== 
    ==14464== Conditional jump or move depends on uninitialised value(s)
    ==14464==    at 0x6C27DAC: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)
    ==14464==  Uninitialised value was created by a stack allocation
    ==14464==    at 0x6C27CDC: vchiq_lib_init (in /opt/vc/lib/libvchiq_arm.so)

Looking at the code at https://github.com/raspberrypi/userland/blob/a1b89e91f393c7134b4cdc36431f863bb3333163/interface/vchiq_arm/vchiq_lib.c there seem to be a number of uninitialised values on the stack.

It's not clear whether the values are tripping valgrind as false positives, however the code would be cleaner if the state was explicitly declared beforehand.

Example:

       static int mutex_initialised = 0;
       static VCOS_MUTEX_T vchiq_lib_mutex;
       VCHIQ_INSTANCE_T instance = &vchiq_instance;
    
       vcos_global_lock();
       if (!mutex_initialised)
       {
          vcos_mutex_create(&vchiq_lib_mutex, "vchiq-init");
    
          vcos_log_set_level( &vchiq_lib_log_category, vchiq_default_lib_log_level );
          vcos_log_register( "vchiq_lib", &vchiq_lib_log_category );
    
          mutex_initialised = 1;
       }
       vcos_global_unlock();

vchiq_lib_mutex looks like it could pass the above uninitialised, when in theory it can't - ideally vchiq_lib_mutex should be explicitly initialised so that if something does go haywire it will fail clean.

In this case "config" is left uninitialised:

             VCHIQ_GET_CONFIG_T args;
             VCHIQ_CONFIG_T config;
             int ret;
             args.config_size = sizeof(config);
             args.pconfig = &config;

pelwell, 2017-05-24:
Please explain systematically how we would end up using a variable in an uninitialised state.

minfrin, 2017-05-24:
Take a look at the vchiq_lib_mutex variable. This variable is initialised in the "if (!mutex_initialised)" path, and is left uninitialised in the else path.

Obviously you and I can see that mutex_initialised is set inside that code and so the "else" path is avoided, but valgrind can't see that, and so gives a warning.

Remember it is not enough to have code that is bug free - you also need code that is not going to trip up the debuggers, and so hide other bugs.

Valgrind complained about three uninitalised values. Was this the one above repeated three times, or are there three separate instances? Easy to tell - fix the false positives, and if you're left with nothing you're fine. If you've fixed all the false positives and the warnings remain, then you have a bug that is hiding among these warnings.

pelwell, 2017-05-24:
Changing code that works to keep tools happy isn't high on our list of priorities. If you want this fixed soon, change the code to your satisfaction and submit a Pull Request.

popcornmix, 2017-05-24:
vchiq_lib_mutex is static and therefore initialised to zero by definition. That is not the problem.

I suspect the report may actually be from config later in the function.
That is set from an ioctl which I don't think valgrind handles:
https://stackoverflow.com/questions/3233174/does-valgrind-track-memory-initialization-through-drivers

JamesH65, 2017-12-05:
Should this be flagged as a non-issue or wont fix and closed?

popcornmix, 2017-12-05:
Without further evidence I believe it is a non-issue (i.e. a spurious report from valgrind).

closed 2017-12-05 by JamesH65