raspberrypi/userland #399: Add support for GL_EXT_bgra

riataman, 2017-05-28:
The raspberry pi GLES implementation includes support for GL_EXT_texture_format_BGRA8888, which adds the BGRA_EXT format. 

It would be nice if it also supported the GL_EXT_bgra extension which in addition to BGRA_EXT  also adds the BGR_EXT format (no alpha channel).

I looked at the little userland code and seems like this is all handled in the firmware.

I was needing this to avoid converting OpenCV camera captures on the client side (as they are returned in BGR format).

Hopefully this wouldn't be so much code.

popcornmix, 2017-05-30:
The firmware side GL driver is being phased out so there will be no new features.
I'd suggest you look into the experimental arm side GL driver to see if it supports what you need.
Search for  ``dtoverlay=vc4_kms_v3d`` and/or ``dtoverlay=vc4_fkms_v3d`` for info.

If not then bug reports and feature requests are handled here:
https://github.com/anholt/linux/issues/

JamesH65, 2017-12-05:
Closing as a wont fix.

closed 2017-12-05 by JamesH65