raspberrypi/userland #400: About exposure mode and ISO

tpengti, 2017-06-05:
When setting off auto exposure, adjusting ISO is invalid,Which is converted to manual exposure mode
ISO is valid only when exposure mode is on，adjusting the shutter seems to be ineffective

What is the relationship between ISO and analog gain and digital gain, I have to modify the userland, add the analog gain and the digital gain of the parameter adjustment, but the prompt function is not implemented.

Looking forward to your reply, thank you very much.

6by9, 2017-06-09:
> When setting exposure=off, adjusting ISO is invalid 

Correct. ISO setup is actually done as a slightly modified auto exposure mode with fixed gain settings.
You should be able to set a manual shutter speed even if exposure mode auto and ISO specified. Only ISO 100, 200, 400, and 800 are supported.

> What is the relationship between ISO and analog gain and digital gain

ISO = analogue gain * digital gain * 100 / gain_for_iso_100.
On OV5647 gain_for_iso_100 is x1. On IMX219 they it was actually calibrated and is x1.84.

> I have to modify the userland, add the analog gain and the digital gain of the parameter adjustment, but the prompt function is not implemented.

No, as discussed on the forums numerous times, there is no direct user control of analogue and digital gains. I had looked in to it and it proved pretty awkward to achieve.

> Whether you can open the Gamma parameter?

Unlikely in the main camera stack.
There are the discussions on the forums over capturing the raw Bayer images and processing those directly - https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=109137.
There is also then the ISP component which is still alpha but should eventually allow you greater control of parameters, however you as the user also then have to take full control over AWB, AE, AGC, lens shading, gamma, etc. If you want the lowest level of control, then that is probably the route you should be investigating.

tpengti, 2017-06-13:
thank you for your reply
Why manual exposure mode, can not adjust ISO?

6by9, 2017-06-13:
> Why manual exposure mode, can not adjust ISO?

It's jsut the way it evolved.
Consider exposure=off to be a lock of all current settings, therefore nothing will change.

Setting ```MMAL_PARAM_EXPOSUREMODE_AUTO``` and ```MMAL_PARAMETER_SHUTTER_SPEED``` to a non-zero value will disable automatic exposure control whilst leaving the automatic gain control active. You can then set your desired ISO value via ```MMAL_PARAMETER_ISO```.

tpengti, 2017-06-14:
@6by9 
I have set Setting MMAL_PARAM_EXPOSUREMODE_AUTO and MMAL_PARAMETER_SHUTTER_SPEED to a non-zero value, but when the ambient light changes, the camera will automatically dim.

As you said (Setting MMAL_PARAM_EXPOSUREMODE_AUTO and MMAL_PARAMETER_SHUTTER_SPEED to a non-zero value will disable automatic exposure control), It should not automatically adjust the light

6by9, 2017-06-14:
Read what I said:

> Setting ```MMAL_PARAM_EXPOSUREMODE_AUTO``` and ```MMAL_PARAMETER_SHUTTER_SPEED``` to a non-zero value will disable automatic exposure control **whilst leaving the automatic gain control active**

It should be solely using gain control to vary the scene brightness.
If you then specify a value via ```MMAL_PARAMETER_ISO``` you will be locking the gain too.

tpengti, 2017-06-15:
@6by9 Thank you very much for your help！
Program running process, can not change the exposure mode and exposure compensation, need to restart the camera, the parameters take effect

JamesH65, 2017-06-15:
Sounds like you have this working, so will close.

For future reference, this sort of question will be better asked on the Raspberry Pi forums, where many more people will be able to see it and answer. This github tracker is really for reporting bugs in the userland code, not usage questions.

closed 2017-06-15 by JamesH65