raspberrypi/userland #401: tvservice -M stopped to works on raspbian jessie 8 pixel?

matteoraggi, 2017-06-08:
Hi, as well documented here https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=184720&p=1173006#p1173006 in may 2017 I made some tets with tvservice and it was looking to works well. With the commands `-p -o -s -M -m -h` , was allright.
Then the first of june I made only this change to the os: sudo apt-get update && sudo apt-get upgrade
and today I find that all commands are working well, apart this one `tvsercie -M` that is answering only in this way and no more:
`Starting to monitor for HDMI events`. 
Instead in may if I was pressing the enter button, then it was showing infos like them on the screen:
```
[I] HDMI cable is unplugged
[I] HDMI is attached
```
Now instead it happen nothing, what should I do to make i works?

pelwell, 2017-06-08:
What is in your config.txt and cmdline.txt?

matteoraggi, 2017-06-16:
any news/helps?

matteoraggi, 2017-06-18:
Just now I tested `tvservice -M` with a fresh install of [RASPBIAN JESSIE WITH PIXEL](https://www.raspberrypi.org/downloads/raspbian/) without updates, just installing dataplicty and here is the behaving:
`[I] HDMI cable is unplugged ` 

> when I unplug by hand the hdmi cable

`[I] HDMI is attached` 

> when I plug by hand the hdmi cable

But if I send the tv in standby or ON it don't find changes, in the past instead it was working, right?

JamesH65, 2018-07-02:
@matteoraggi Is this still an issue in the latest version of Raspbian?

matteoraggi, 2018-07-05:
I don't know, I should restart this type of tests minimum from 1-2 weeks or month..