raspberrypi/userland #404: Add support for circular buffer in raspivid witout closing when trigering USR1

MeAntonio, 2017-06-21:
Would be great to make that raspivid doesn't stop when triggering -USR1 signal in circular buffer mode, only taking t seconds of video and write it into file video%04d.h264 or directly to stdout. It would be great to trigger it more than once without spending time on initializing camera every time.

I'm looking into solution.

Tonchyk, 2017-06-29:
Here is my solution: (I'm also meAntonio)

Added 1 more option -cm that activates saving circular buffer into file like this_%03d.h264 which for every -USR1 signal saves circular buffer into file and then opens new file. Property -cm should be used with -c. 
Also when setting -t under 2 sec You should set -g to 2-5 to work properly.

Here is the example of usage:
raspivid -c -cm -n -o name_of_video_%05d.h264 -t 1000 -s -g 3 -b 3000000
- creates 3mbps video file that lasts for approximately 1 second with 3 p-frames between i-frame

c - activates circular buffer 
cm - disables exiting when sending a signal
n - no preview
o - output 
name_of_video_%05d.h264 - name of saving files (name_of_video_00000.h264, name_of_video_00001...)

You send a signal with:
kill -USR1 <pid> //you can get the pid with "pidof raspivid" command.

Still not working 100% percent, after some time ON it starts throwing mmal error Error in frame list which I haven't fully understood yet.

https://github.com/Tonchyk/raspivid-multitriger-circular-buffer

JamesH65, 2017-12-05:
It's much easier for us to see what you have done if you submit a pull request (we get all of githubs code comparisons to play with). So if you are still working on this, that might be a good approach. If you are not working on it, can you close the issue?

Tonchyk, 2017-12-05:
Will submit soon, got it working.

napisala je:

JamesH65, 2018-07-02:
@Tonchyk Was this PR ever submitted?

syadnom, 2018-07-16:
yeah, would really like to use this...

JamesH65, 2018-07-16:
Happy to assess any PR's that turn up.

jimbo1969, 2018-09-25:
This is exactly what I'm looking for as well.  I'm wondering if it is already implemented somehow/somewhere.  My apologies if it is obvious.  Is any other follower of this thread aware of any publicly-available implementation of this functionality?  I don't know how to tell which GitHub branches are really making their way back into the "official" raspivid/picamera code, so I feel like I'm barking up a bunch of random wrong trees.

I'm looking to repeatedly trigger the storing of ~5 sec video clips to a large storage device for postprocessing.  My trigger will not occur until halfway-through the clip, and I might occasionally have one immediately (~1 sec) following another.

Thanks!