raspberrypi/userland #405: MMAL camera image overlay Example?

AlexanderShniperson, 2017-06-29:
Hi,
can some one make an example of creation overlay image over camera preview, like it does in [Python Picamera](https://github.com/waveform80/picamera)?
p.s. it is hard to understand Python source code
p.p.s. i am from Java world and looking for simple solution
thanks.

6by9, 2017-06-29:
As from https://github.com/raspberrypi/firmware/issues/832:

> Overlay over preview (not recording it) is just creating a MMAL video_render component, setting the display rectangle and layer (MMAL_PARAMETER_DISPLAYREGION) appropriately, and sending it a buffer with your image in it.

None of the userland code is in Java, and is very unlikely ever to be.

Trivial example app pushed to https://github.com/6by9/mmal_render_example. I'll look at pulling it into userland at some point in the future.

AlexanderShniperson, 2017-06-29:
you understood me wrong, i dont mean example on the Java.
[Java MMAL implementation is exitst](https://github.com/caprica/picam) but [author have a problem](https://github.com/caprica/picam/issues/2) with `mmal_port_send_buffer` where last buffer of capture doesn't came in about 10% of times (as example 10 times from 100)

AlexanderShniperson, 2017-06-29:
@6by9 thanks for example, is there a plans to modify it to take some picture from the disk and place it as overlay?

6by9, 2017-06-29:
> Java MMAL implementation is exitst but author have a problem with mmal_port_send_buffer where last buffer of capture doesn't came in about 10% of times (as example 10 times from 100)

How does that relate to wanting overlays?
If you/@caprica want assistance with that then create a new issue with **simple** steps to reproduce it, and preferably logs, then we may be able to help. Having just read through that issue there is no detail to work from, and spurious references to unrelated issues (stereoscopic captures with burst mode, or video_decode stalling?!) and APIs (DispmanX).
There isn't even a statement of which image format is being used. A quoted error message referencing PNG is about the most definitive. PNG has no hardware acceleration on the GPU, and being RGB based instead of YUV requires image conversions to be done. 
No definitive statement of resolution either. One passing code snippet implying 1920x1080, but is it only that resolution? If so then the fact 1080 is not a multiple of 16 would be the smoking gun to me.

>  is there a plans to modify it to take some picture from the disk and place it as overlay?

In a word, no.
If you want to do that then it makes some sense to tunnel image_decode to video_render, but there are also so many image decode libraries that anyone can tackle decoding an image into the buffer.

Please note that all the limitations on padding as documented at https://picamera.readthedocs.io/en/latest/recipes1.html#overlaying-images-on-the-preview apply to this too (unsurprising seeing as they use the same rendering engine).

AlexanderShniperson, 2017-06-29:
@6by9 ok thanks for explanation, we will open an new issue and give you detailed info.
about problem of sending buffer: what kind of logs do you want to see?

6by9, 2017-06-29:
Anything and everything that you think might be relevant. Start with:

- Enable MMAL logging (```export VC_LOGLEVEL="mmal:trace"``` before you run your app) and you should get a large amount of stuff printed to stderr associated with your app.
- ```sudo vcdbg log msg``` for the appropriate time period (the first number is a timestamp in microseconds).
- Add ```start_debug=1``` to ```/boot/config.txt```, and then the output from ```sudo vcdbg log assert```

6by9, 2017-11-17:
Closing due to inactivity.

closed 2017-11-17 by 6by9

dumski, 2020-09-05:
Hi, soryy for bump this but I cannot find an information in the MMAL documentation and forums, how to render multiple layers on render casomponent? I tried to allocate more input ports using mmal_ports_alloc, no luck. Ports are succesfully allocated, but cannot commit format on additional ports.

Is multiple input ports a correct approach to multilayer render on render MMAL component? 
Cheers

6by9, 2020-09-05:
> But my second question still remains: is this (creating component for every overlay layer) a good approach?

It's the only approach. video_render has a single input port and a single set of parameters as to where the overlay is to be positioned.