raspberrypi/userland #409: ARM64: EGL/GLES2 Libraries unable to compile

raidensnake, 2017-07-14:
I've tried compiling EGL/GLES2 for SDL but it won't compile. I mainly get errors relating to Khronos.

yfang1644, 2017-08-26:
hi!
I tried compiling this library in arm64. But it seems not working. Followings are what I did:

1. change cross-compiler in arm-linux-gnueabihf.cmake to `aarch64-linux-gcc/g++
2. remove khrn_int_hash_asm.s in CMakelists.txt since it is written for arm32
3. change sizeof (some pointers) in khrn_client_check_types.h from '4' to '8'
4. change some pointers from 'int' to 'long' in .c sources
5. some log print from '%u' to '%lu'. I think this should not so important, only for syntax

After many times trying, It is nearly successfully compiled except some warnings. 
EGL works, but audios may not work. I'm not sure if the problem comes from kernel or userland.
The kernel I'm running is linux-rpi-4.11-y
......

raidensnake, 2017-09-03:
Any idea what files and pointers to change?

Also the sound works with a pimoroni PHAT DAC on 4.11 or the onboard hdmi sound works on 4.13.

JamesH65, 2017-12-05:
We are not planning a 64 bit version of this stuff, its a lot of work and the new ARM side driver VC4 is now available, albeit not yet the default.

If other people are happy to do the work then we would certainly look at a PR.

Closing as wont fix.

closed 2017-12-05 by JamesH65