raspberrypi/userland #411: OMX.Broadcom.clock fails to transition to idle

estobbart, 2017-07-22:
Working on an OMX player, I've loaded the clock, tunneled it to other components. When I request transition to idle the callback never fires and the component is still in it's loaded state. I had accidentally tried transitioning it to executing and it failed with a OMX error insufficient resources.
The component only has output ports, what resources is it failing to acquire?
All other components seem to successfully transition to idle and then to executing except for the clock.

6by9, 2017-07-25:
Have you disabled all unused output ports before trying to enable the component? If not then IIRC they'll be waiting for a full complement of buffers to be allocated before the transition completes.

6by9, 2017-11-17:
If you still have an issue then please respond to the questions asked, otherwise the issue will get closed.

JamesH65, 2018-07-02:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-07-02 by JamesH65

estobbart, 2018-07-02:
Apologies for not following up, disabling all the output ports (not being used) was the issue. Thanks!