raspberrypi/userland #412: ISP differences about IMX219 and OV5647

tpengti, 2017-07-25:
When using the OV5647, I can setting exposure=off.
but when using the IMX219, when exposure=off or exposure=auto， they have the same effect, the feeling exposure=off  is not valid for IMX219
Looking forward to your reply！

6by9, 2017-07-25:
You'll need to give some more details of your setup as it works fine for me.

```raspistill -set -ex off -ss 20000```
and the settings values logged all come back the same exposure time and gains (add ```-awb off -awbg 1.0,1.0``` to shut up the changes due to AWB).
```
mmal: Exposure now 19953, analog gain 256/256, digital gain 256/256
mmal: AWB R=256/256, B=256/256
```

tpengti, 2017-07-28:
pi@raspberrypi:~ $ raspistill -ex off -awb off -ISO 100 -cfx 128:128 -set
mmal: Exposure now 19991, analog gain 256/256, digital gain 256/256
mmal: AWB R=0/256, B=0/256
pi@raspberrypi:~ $ raspistill -ex off -awb off -ISO 800 -cfx 128:128 -set
mmal: Exposure now 19991, analog gain 256/256, digital gain 256/256
mmal: AWB R=0/256, B=0/256

When I set up the ISO 100 or 800, the analog gain and the digital gain value did not change.

ethanol100, 2017-07-28:
Does the behaviour change, if you specify the awb gains(-awbg 1,1)? Your awb gains are 0 and I don't know if this will influence the rest. 
Can you try the simplest set of parameter only specifying -ex off -ISO -v?

tpengti, 2017-07-28:
I have tried to set it up. -ex off -ISO 100 (or 800), but the ISO value will not change, all display 256/256.
I would like to know, in the case of -ex off, how to set the analog gain and digital gain, that is, to make ISO effective.

ethanol100, 2017-07-28:
Yes, sorry. I think what you want to achieve is not possible. Setting -ex off in raspistill/raspivid will always use analog gain 255/255 and digial gain 255/255. You can try to only specify awb gains, setting shutter time and ISO. For the V2 camera this should give relative constant values for the gains. I think this is the only way to get different values for the gains. The problem is that the camera starts with -ex off and will not have any algorithm running to calculate the gains. The firmware then sets the gains from initial 0 to 1. The auto exposure algorithm would need to run for some frames to set up the correct gains.Then you could fix them to the current values setting the exposure mode to off. But this is not an option or raspistill/raspivid and will not let you choose specific values for the gains.

tpengti, 2017-07-28:
@6by9 
https://github.com/raspberrypi/userland/issues/413
As I have this problem, the metering system is sometimes incorrect, my application scene is to detect the bar code, so I would like to set the -ex off the situation, the program according to the bar code brightness to set the shutter and gain, do not want other areas to lead to bar code Burst or too dark

6by9, 2017-10-10:
@tpengti I believe all your questions have been answered, and you can now set your own explicit analogue and digital gains (see https://github.com/raspberrypi/userland/pull/430).

Please close if you have no further questions on this.

6by9, 2017-11-17:
Closing due to inactivity.

closed 2017-11-17 by 6by9