raspberrypi/userland #413: Manual exposure algorithm, manually obtain the target area of the brightness information to set the camera parameters

tpengti, 2017-07-25:
Sometimes the camera's metering is not allowed, if the camera according to the automatic exposure, the screen will be excessive or inadequate, such as night and light contrast contrast is very large, manual exposure can get their own combination of exposure effects
Raspberry faction can set the manual exposure (manually set the shutter, gain), these parameters are not affected by external lighting.
Thanks for  your work！

6by9, 2017-07-25:
I'm not clear what you're asking for here.

Manually setting the shutter speed can already be done (-ss option in raspistill/vid).
Analogue and digital gain can't be explicitly set, but the combined value can be set via the ISO parameter.

Serveurperso, 2017-09-19:
I can't set the - ISO parameters, it's always ignored even with - ex off

raspivid -fps 25 -w 720 -h 576 -ex off -ISO 800 -cd MJPEG -o /var/www/temp/video.mjpg

tested many combinations of - ex off/auto/night without success

6by9, 2017-10-10:
@tpengti https://github.com/raspberrypi/userland/pull/430 is now merged which allows you to manually set both analogue and digital gains. Manual shutter speed and white balance gains has been possible for a while.
I think that is what you were requesting although you never confirmed it, so I'm closing this issue

closed 2017-10-10 by 6by9