raspberrypi/userland #414: The exposure Settings of the camera (the company needs more)

ptlinus, 2017-07-30:
The raspberry pie is a very good product, and our company has a lot of purchasing every month, but we want to solve this problem.

I would like to know, in the case of -ex off, how to set the analog gain and digital gain, that is, to make ISO effective.(If you can't set it right now, can you modify this situation)

Thank you very much. It will be a decisive help for our products.

Serveurperso, 2017-09-19:
same problem here: for a home project. I can't set the - ISO parameters, it's always ignored even with - ex off

raspivid -fps 25 -w 720 -h 576 -ex off -ISO 800 -cd MJPEG -o /var/www/temp/video.mjpg

tested many combinations of - ex off/auto/night without success

6by9, 2017-10-10:
Fixed.
Commit https://github.com/raspberrypi/userland/commit/212184f0f1fdf76eff31e3875fdeb2b7980cd5cb adds ```-ag <float>``` and ```-dg <float>``` options to the raspicam apps to fix the analogue and digital gains respectively.
It is using part of the AE/AGC algorithm, so do NOT set ```-ex off``` with these options.

If there is something specific that you find doesn't work with it, then please raise a new issue.

closed 2017-10-10 by 6by9