raspberrypi/userland #415: RaspiCamControl?

rodizio1, 2017-08-15:
I have seen sources of a tool called "RaspiCamControl" in the userland directory (https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/raspicam). 

Thought I'd try it out, cloned the userland repository, ran the buildme script, but the Raspicamcontrol tool did not get built.

Searched the web for it, couldn't find anything useful on how to build it or what it does.

Could somebody please shed some light into this? How can it be built? Is my assumption correct, that it is something similar to v4lcontrol, i.e. changing camera settings while the camera is already running?

JamesH65, 2017-08-16:
> master/host_applications/linux/apps/raspicam).

That's just one of the source files used by the camera apps, not a separate
tool.

rodizio1, 2017-08-16:
D'ohh. Thanks for the info :)

closed 2017-08-16 by rodizio1