raspberrypi/userland #416: raspivid - video streaming

plumpsack, 2017-08-20:
`raspivid -l -o tcp://0.0.0.0:3333`

(raspivid -l -o udp://0.0.0.0:3333 , not working)

is only working for "one shot".

If one client is/was connected, raspivid is closing its service.

There is in the moment, since a long time, only one way to enable a listening service for the video stream like:

```
sudo su
apt-get install vlc
cp /usr/bin/vlc /usr/bin/vlc_backup
sed -i 's/geteuid/getppid/' /usr/bin/vlc
```

`vim /etc/rc.local`

```
##################################
raspivid -t 0 -w 960 -h 540 -fps 0 -br 50 -b 1000000 -ex night -a 8 -a 1024 -a "CAM04 %Y-%m-%d %X" -o - | cvlc 
stream:///dev/stdin --sout '#standard{access=http,mux=ts,dst=:3333/stream1.h264}' :demux=h264

# -fps 0 for night view (bug ?)
# use http://rpi-address:3333/stream1.h264 to watch the video stream on the client

##################################
exit 0
```

Could you please fix that?

JamesH65, 2018-12-03:
@plumpsack Is this still an issue? I'm not entirely sure what the problem is from your description.

plumpsack, 2018-12-04:
Hi, same problem with the new "version", version is internal the same "raspivid Camera App v1.3.12"

On the raspberry i start 

raspivid -l -t 0 -w 640 -h 480 -o tcp://0.0.0.0:3333

(raspivid -l -w 640 -h 480 -o tcp://0.0.0.0:3333 is not working for a video, just a one shot picture)

client: ffplay tcp://raspberrypi3:3333
video ok
quit client

answer on pi:
Waiting for a TCP connection on 0.0.0.0:3333...Client connected from 192.168.xxx.xxx:45846
mmal: Failed to write buffer data (3746 from 51416)- aborting

I am on 4.14.84-v7+ #1169 SMP Thu Nov 29 16:20:43 GMT 2018 armv7l GNU/Linux, Stretch

Any idea?

`file /opt/vc/bin/raspivid
/opt/vc/bin/raspivid: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=d5f722f05133653e89c11e40fa60fb2f7fb54c15, not stripped

ls -la /opt/vc/bin/raspivid
-rwxr-xr-x 1 root root 72600 Dez  4 12:24 /opt/vc/bin/raspivid`

plumpsack, 2018-12-04:
@JamesH65,

what did you tried to change?

A) the re-connect problem with raspivid
or
B) the problem with fps in night view?

6by9, 2018-12-04:
a) raspivid is not a fully fledged server (it's a heavily hacked demo app) so I'm not sure how practical it is to have it reopen a listening connection on disconnect.

b) What problem with night view? There only appears to be a half reference in your script, no description of what you mean.
Setting fps to 0 means that you allow the frame rate to vary based on the desired exposure time. That's not a bug.

plumpsack, 2018-12-04:
@6by9,
have you ever tried the different between fps=0 and fps=15 in night view?
If not, do not give a comment!

fps=0 is not working in the night!

raspivid is, as i said in 2017, not usable for streaming.

Would be nice to get a working version.

I am still using cvlc, as i wrote, that is "nearly" working.

JamesH65, 2018-12-05:
Wow, way to piss off one of the few RP engineer who could actually help you. Nice going.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

6by9, 2018-12-05:
For my own curiousity I did run this (it might have been pink unicorns!). It all seems to be behaving as expected.

```raspivid -l -o udp://0.0.0.0:3333``` is meaningless. It's effectively "listen for a connection on a connectionless protocol" - there is no connect event. 

Yes, the app quits when the client disconnects, or the timeout period expires after the client disconnects. That is consistent with the behaviour on all other modes of operation of raspivid in that it streams to a single file/network address until the timeout expires and a file operation failure (eg SD card full) results in an immediate close.
```raspivid -l -t 0 -w 640 -h 480 -o tcp://0.0.0.0:3333``` is telling it please listen for a connection, and then stream indefinitely to that destination. Omitting ```-t 0``` means stream for the default 5 seconds to that destination. If you destination hasn't started decoding within 5 seconds then that's not the fault of raspivid.
TBH the simplest approach is write an external script that calls raspivid again when it quits. You need to reset the codec anyway to produce a new set of header bytes etc, therefore it's no big deal to reinitialise everything.

Please note that raspivid sends a raw H264 (or MJPEG) elementary stream over the network. It looks like you're telling cvlc to encapsulate it in HTTP. Don't expect trying to decode via http:// to work directly from raspivid as the HTTP headers won't be there. ```vlc tcp/h264://<ip>:<port>``` appears to work correctly, although again it has no time reference and will complain should the inter-frame time exceed some arbitrary threshold.

```-fps 0``` allows auto exposure to adjust the frame rate at will. I've run that and see no issues:
```raspivid -fps 0 -ex night -o tcp://0.0.0.0:3333 -l -v -w 640 -h 480 -b 1000000 -t 0 -fl -set``` on the Pi.
On the receiving Pi minor hacks to hello_video to allow it to receive data from stdin, and ```nc 192.168.2.201 3333  | ./hello_video.bin -``` works fine.
The main issue is latency within Linux as it buffers stdin (and stdout). That's why ```-fl``` has been added to the raspivid command to flush the output pipe after every write.
On hello_video it defaults to 80kB buffers on the input to video_decode, and therefore Linux waits until it has 80kB to return when we call fread. 80kB is quite a lot of data for a compressed stream, so often includes several frames. hello_video decodes and displays the data as it arrives, therefore we see bunching together of motion followed by a delay. Using a smaller buffer size (20kB) improves that, however when the frame rate drops due to dark conditions the latency increases because it takes longer to fill that buffer. Sorry, that's an issue in the receiving app, not raspivid.

Use ```nc``` to pipe to a file and that also produces a totally valid H264 stream. Playback varies in speed as elementary streams have no timestamping information in them.

So I have tried it and found no fault.
Over to you to actually describe what your issue is (as would have been useful in the first place).

JamesH65, 2018-12-12:
@plumpsack Awaiting your input on this before anything else can be progressed.

beta-tester, 2019-03-12:
hello, i just tried to use `raspivid -t 0 -n -l -o tcp://0.0.0.0:1234` on my RPi and on my PC i tried to use VLC with `tcp://192.168.1.100:1234` to open the video stream.
on the RPi i see that the connection is established but on VLC i can not see the video... on the codec tab is nothing listed, on the statistics tab everything is 0 (zreo) except of the media size, that is growing endless and th input-bitrate is at about 17000kb/s.

what is raspivid streaming?

i have also opened a thread [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=235684)

BTW: `raspivid -t 0 -n -l -o udp://0.0.0.0:1234` isn't working at all, `No support for listening in UDP mode`.
and without listening i get:
```
~ $ raspivid -v -t 0 -n -o udp://0.0.0.0:1234

"raspivid" Camera App (commit 7cbfbd38d982 Tainted)

Camera Name ov5647
Width 1920, Height 1080, filename udp://0.0.0.0:1234
Using camera 0, sensor mode 0

GPS output Disabled

bitrate 17000000, framerate 30, time delay 0
H264 Profile high
H264 Level 4
H264 Quantisation level 0, Inline headers No
H264 Fill SPS Timings No
H264 Intra refresh type (null), period -1
H264 Slices 1
Wait method : Capture forever
Initial state 'record'

Preview No, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera video port to encoder input port
Connecting to 0.0.0.0:1234...connected, sending video...
Opening output file "0.0.0.0:1234"
Enabling encoder output port
Starting video capture
mmal: Failed to write buffer data (692 from 2587)- aborting
Finished capture
Closing down
Close down completed, all components disconnected, disabled and destroyed
```

JamesH65, 2019-03-13:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2019-03-13 by JamesH65

kevr, 2020-10-28:
What exactly is the reason that `raspivid` isn't implementing a reusable server?

JamesH65, 2020-10-28:
All the camera apps were originally written as demo code, simply taking images/videos, with the intention that people would modify it for their own purposes. However over time more and more stuff has been added, so they are rather a mismatch of features nowadays.

We don't have time to redo it, and libcamera will be our future focus anyway.

kevr, 2020-10-28:
Gotcha, thanks for the reply!