raspberrypi/userland #421: Raspivid causes hard crash when using hflip/vflip and hdmi

sej7278, 2017-09-11:
See also [forum post](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=192933)

Running the following.....

```
raspivid -t 0 -w 1920 -h 1080 --hflip --vflip -fps 30 -b 2000000 -o - | \
gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=1 pt=96 ! \
gdppay ! tcpserversink host=desktop.lan port=5000
```

....causes my Pi ZeroW to lock up. 

Once it didn't totally kill the machine and I managed to get some [dmesg output](https://gist.github.com/sej7278/9783d0e3c6f9287d4047a86d45e2b3b0), although the raspivid output had stopped it worked again on restarting the process.

Once it logged a systemd error "Internal error: Oops: 817 [1] ARM", but usually it just totally crashes the machine, a P6 reset doesn't work, only removing and re-applying usb power will reboot it.

After some experimentation it would seem that it doesn't crash if I add the -n flag to disable preview to HDMI (i don't have a monitor plugged in).

It also doesn't crash if you remove the --hflip and --vflip flags, so it would seem rotating and outputting to HDMI and wifi is too much for a ZeroW

Its not a power issue as I've used two supplies and it never draws more than 0.6A or dips below 5.0v

This is a fully updated Stretch with the 4.9.41 kernel

6by9, 2017-09-11:
H and V flips are done on the sensor itself by reading out the sensor array in a modified order. No extra GPU load at all.
Rotate (specifically transpose) is done on the GPU (and is expensive in terms of performance and SDRAM bandwidth), but I see no evidence that you're doing a transpose.

As DirkS suggested on the forum thread, break it down.
Does ```raspivid -t 0 -w 1920 -h 1080 --hflip --vflip -fps 30 -b 2000000 -o /dev/null``` crash?
Seeing as you've managed to crash the kernel, I'd suspect something further downstream in your GStreamer pipe (probably 

BTW 2Mbit/s is a pretty low bitrate for 1080P30. For reasonable quality I'd be expecting >8Mbit/s.

sej7278, 2017-09-11:
ah no, i can't get it to crash when its going to /dev/null instead of gstreamer. going to try plugging a monitor in too

6by9, 2017-09-11:
Next step then is to start adding in the GStreamer stages and terminating the pipe in a ```fakesink```. That should happily swallow anything.

sej7278, 2017-09-11:
i'll give it a go, not an expert on gstreamer, not sure if it needs a client connection to crash. i wonder if vlc would be better than gstreamer?

i plugged in a monitor and keyboard and when i changed the bitrate to 8mbps it eventually dropped the stream and i couldn't ssh in, but the screen was still updating so i guess it just overwhelmed the wifi, when i did a P6 reset there was some stack trace on there talking about iptables being linked in, so i guess it was the wifi driver that crashed.

i managed to make it crash using my original command with the gstreamer and the screen even stopped updating so it definitely crashes totally, its not just wifi.

6by9, 2017-09-11:
Each of the !'s is denotes a new stage in the processing pipe. You should be able to put ```fakesink``` after any of them and have a working pipeline.

Start with ```raspivid -t 0 -w 1920 -h 1080 --hflip --vflip -fps 30 -b 2000000 -o - | \
gst-launch-1.0 -v fdsrc ! h264parse ! fakesink```
If that works then increase it to ```raspivid -t 0 -w 1920 -h 1080 --hflip --vflip -fps 30 -b 2000000 -o - | \
gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=1 pt=96 ! fakesink```
and continue up to your full pipeline.

sej7278, 2017-09-11:
thanks, will do

sej7278, 2017-09-11:
the only way i can crash it is with tcpserversink **and a connected client.** it doesn't seem to crash just running the server, so i assume its the wifi load or something....?

i noticed the client crashed, ssh connection froze and the screen got laggy. P6-reset doesn't work but does exit raspivid back to the console and shows ```"Internal error: Oops: 817 [1] ARM"``` and the kernel null pointer dereference from the above dmesg gist.

6by9, 2017-09-11:
So I'd suggest you test with an alternate USB Wifi or ethernet dongle to narrow down what is causing the failure.
There were a couple of fixes to the Wifi firmware and driver recently, but I thought those were in before 4.9.41+ (around 8th August).

sej7278, 2017-09-11:
not sure i'm up to debugging wifi drivers!

similar issue perhaps - zerow wifi dies when streaming camera: https://github.com/raspberrypi/linux/issues/1342

6by9, 2017-09-11:
I'm not asking you to debug the driver, but pinpoint where the issue is.

As I had said, there were changes pushed recently for the onboard Wifi firmware and driver (it's common between PiZero and 3). Some are still reporting issues, but at the moment you haven't pinned down that that is the problem in your specific case.

sej7278, 2017-09-11:
ok, i managed to make it crash using a usb ethernet dongle: [gist](https://gist.github.com/sej7278/3f62643b729c82a20e10da7db7b5285e)

seems to be another "pi zero locks up" thread on the [forum](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=158331) with the same arm oops when running vnc

sej7278, 2017-09-11:
oddly enough i cannot get an old zero 1.3 to crash, even when streaming at 8mbps over usb wifi dongle. i did notice it ran about 8c cooler than the zerow, so i guess the wifi chip gets hot.

going to try swapping the sdcards next, i'm really lost at this point.

sej7278, 2017-09-12:
Not sure the title is even right now as i could crash over ethernet.

I'd like to know if someone else could reproduce.

6by9, 2017-09-12:
Feel free to change the title yourself then. It was totally incorrect as there is no issue in raspivid.

The Github repos are frequented by very few people, and your issue is looking to be kernel related rather than userspace - that would be https://github.com/raspberrypi/linux/ rather than here.
I'd update your forum thread and ask there, or start a follow-on thread in an appropriate forum ("Networking and servers" probably).

If you can test on a second Pi0W then it'd be worth it.
This issue sounds like a hardware problem, so I wonder if your Pi is struggling to achieve the normal 1GHz that the Pi0 runs at. You could try underclocking it too to see if that helps.

sej7278, 2017-09-12:
ok, thanks for the advice. i'll try underclocking and a new thread in the kernel forum.

i'm not prepared to buy another zerow as i've spent too much on this failed project already (zerow has been pretty flaky - like the gpio-shutdown-overlay rarely works.) i was hoping a foundation staffer could try to reproduce.

i might just use a monitor and forget streaming, in which case i could have just used a zero1.3 or something off-the-shelf.

6by9, 2017-09-12:
> i was hoping a foundation staffer could try to reproduce.

The company would go bust in 5 minutes flat if it tried to replicate every single failure reported, particularly as they are often poorly described, or likely to be specific to the reporter's hardware. How much time do you think the profit margin on a Pi0W would cover salary and associated costs for?
We're happy to help where applicable, and will react where there is a specific problem, but forum and Github support is normally being done whilst waiting on other things - we're all working on other stuff too.

sej7278, 2017-09-12:
sorry i didn't mean for you to test it necessarily, just thought there may be a process for raising issues with the foundation (how else would anything get fixed?)

anyway, i've underclocked to 800mhz and it seems stable, still much hotter than my zero1.3 but no cooler than at 1ghz, so i guess its not a temperature problem.

no idea what it could be really, just something in the sbc that doesn't like 1ghz, which is disappointing as that's the speed they're marketed at. i do recall they were initially shipped at 700mhz like the earlier pi's, is 1ghz actually an overclock?

JamesH65, 2017-09-13:
They are sold at 1Ghz. If your's isn't working correctly at that speed (its very rare but possible)I suggest either getting a refund (is it worth it for $5? You might find a use for it elsewhere) or simply getting a new one.

sej7278, 2017-09-13:
yes i can't be faffing around for a refund (although its actually $16 with postage and tax, not $5) but really just wanted assurance that if i buy another it won't have the same problem. so they're not overclocked 700mhz chips or anything? might just be a bad one?

6by9, 2017-09-13:
There's no "binning" of devices to be sold as running at differing clock speeds in the way Intel and similar will do. It's the same chip put into all Zero, ZeroW, B+, and A+ boards.

The original designed spec was 700MHz, but production and yield improvements has actually shown that almost all will run at 1GHz or more (some people overclocked early Pi's up to 1.4GHz IIRC). There will be the odd one that comes off the production line that will run happily at the original 700MHz spec, but not stable at higher clock rates. They are the exception, and unfortunately there isn't an easy way to weed them out in testing. It seems you've got one of the weaker ones and we can only apologise for that.

sej7278, 2017-09-13:
ok no problem, i'll get another and see how it goes. i'll close the issue as its filed under the wrong repo anyway.

closed 2017-09-13 by sej7278

thebeachtoday, 2017-12-07:
I had a similar situation with the RPi3.  Need some ideas how to troubleshoot further...

Issue: client connected to WiFi stream causes severe WiFi performance issues.

Test Case: Two RPis [with identical SD cards], 1 meter apart.  When I run an iperf on them, I get around 40Mbits/s.  Turning on UV4L streaming does nothing to impact WiFi bandwidth.  With UV4L running, as soon as a client connects to view the stream via local browser [e.g. http://192.168.0.27], the bandwidth drops to 5Mbits/s.  As soon as the client drops, the bandwidth jumps back up to 40Mbits/s.  

To take this further, when I move the RPis 25 meters apart, the bandwidth is about 20Mbits/s.  As soon as the client connects, the WiFi gets hosed, and I have to do a cold boot to bring it back - similar to what the OP was seeing with Pi Zero.

I'm using node js to serve the stream via an HTML page.  I have created other non-streaming pages, and visiting those pages do NOT have an impact on WiFi bandwidth.

When issuing a top command after the client is connected to the streaming HTML page, UV4L jumps up in memory usage, but only to 10%.  All other CPU usage and Memory Usage stays about the same.  I do not think this resource uptick has anything to do with it per se, so I tested "other" physical items...

When an HDMI monitor is connected, the same drop in bandwidth is experienced, but nothing eats up resources via top.  I've tried different [and bought new] power supplies, and I also confirm 2A out with meter.  I imagine power usage has a play in this.  It seems as if WiFi is deprioritized vs. other activities taking place???  

Any ideas how to troubleshoot further?

maciek01, 2020-02-03:
same problem here on 2 different RPIZWs - i loose them all after 10-15 mins streaming via gst-launch-1.0 - streming out via wlan0 or ppp0 (LTE modem) - same issue.