raspberrypi/userland #422: The difference between EV and ISO

tpengti, 2017-09-13:
I know the principles of ISO："ISO = analogue gain * digital gain * 100 / gain_for_iso_100."
But not clear EV, this parameter is to adjust which parameters of the camera, is shutter?  analogue gain?digital gain?

Looking forward to your reply！thanks

6by9, 2017-09-13:
https://en.wikipedia.org/wiki/Exposure_compensation and https://en.wikipedia.org/wiki/Exposure_value

Basically EV leaves everything at auto, but adjusts the target for those automatic algorithms.

tpengti, 2017-09-13:
@6by9 Thank you very much for your reply！
E_{v}=A_{v}+T_{v}\,,  
Aperture parameters of the same situation, the raspberry products to send this EV should only be associated with the shutter

6by9, 2017-09-13:
> Aperture parameters of the same situation, the raspberry products to send this EV should only be associated with the shutter

Is that a statement or a question?

The original definitions are from the old film world where you could only change the gain by changing the film for one of a different speed. Most small digital sensors will combine control of shutter and gain to achieve the equivalent behaviour as stopping up or down in the film world.

6by9, 2017-11-17:
Closing due to inactivity

closed 2017-11-17 by 6by9