raspberrypi/userland #424: raspivid MJPEG always too dark, how to push the gain?

Serveurperso, 2017-09-19:
I need >= 25FPS (slower is useless for my home real-time robotics) 

I can't set the - ISO parameters, it's always ignored even with - ex off

raspivid -fps 25 -w 720 -h 576 -ex off -ISO 800 -cd MJPEG -o /var/www/temp/video.mjpg

tested many combinations of - ex off/auto/night without success

How to push up this gain? (noise is not a problem!)

Serveurperso, 2017-09-19:
```
(root|~) vcgencmd version
Sep 12 2017 15:25:20
Copyright (c) 2012 Broadcom
version 3fbb171d1950eb3789eb8e37ba11e01079a37239 (clean) (release)
(root|~)
```
all last available version on raspberry pi 3 (stretch)

JamesH65, 2017-09-19:
Why have you opened three issues for the same problem? It's not going to
make things progress any faster.

Serveurperso, 2017-09-19:
Sorry for this, I opened my own issue because I think my question is simpler :
"how to push the gain and get the noise of the sensor visible on MJPEG video ?"

You can close this one or any duplicate, I am at your disposal for tests.

There is a dedicated raspberry pi 3 24/7 online here - the raspberry pi camera port of my home robot project - where I need more video gain :
http://www.serveurperso.com:8080

To compare, here is the working camera system (my original project) with very sensitive CMOS and HAD2 camera :
https://www.serveurperso.com/?page=robot

(no ads on any URL)

6by9, 2017-09-19:
Remove ```-ex off```. ISO is part of the automatic gain control algorithm, therefore turning that off stops ISO working.
Add ```-set``` and the app will print out the current exposure time, analog and digital gains, and white balance gains. 

Also remember that increasing the gain whilst auto exposure is still running will just result in the exposure time being shortened. To adjust the target brightness of the scene you need to be increasing the EV to alter the target.

Serveurperso, 2017-09-19:
I already tested -ISO 800 with -ev 10 and without -ex
without significative improvements.
I re do this test with -set.
Parameters are displayed on stdout or OSD?

6by9, 2017-09-19:
> Parameters are displayed on stdout or OSD?

stdout (which would have been blatantly obvious if you tried it).

Run ```raspivid -fps 25 -w 720 -h 576 -cd MJPEG -o /var/www/temp/video.mjpg -set```
If the exposure time has hit 39840 usecs and analog gain is at 2521/256 (x9.84) then you have hit the maximum the sensor will do. Unless you can rewrite the laws of physics then that is a hard limitation.
(I'm assuming you're using a V2 IMX219 sensor. The V1 OV5647 has slightly different maximum values).

You can use the -br (brightness) option to crank up the gain later in the processing pipe. Range is 0 to 100.

JamesH65, 2017-09-19:
OV5647 max figures are :

39806us, 2048/256 analog gain.

What specifically are you trying to do, because it all seem to be working
as I would expect?

Serveurperso, 2017-09-19:
I'm away from my raspberry (at work) and there is too light to test for now:D
I run your cmdline tonight. 
I use mjpeg-streamer experimental it give the same parameters as raspivid
I configured like this for now.

```
(root|~/scripts) cat mjpg-streamer.sh
#!/bin/bash

pat=/root/mjpg-streamer/mjpg-streamer-experimental
bin=$pat/mjpg_streamer

i="input_raspicam.so -fps 25 -q 10 -x 720 -y 576 -rot 180 -ex night -ISO 800 -ev 10"
o="output_http.so -p 8080"

LD_LIBRARY_PATH=$pat $bin -i "$i" -o "$o" -b

```
This is bad if the hardware have gain limitation to hide the sensor noise. On many module I can do it.
Add brightness is bad because it's not automatic and white parts of the picture get overexposed on day light
sorry for my English

Serveurperso, 2017-09-19:
I look on Google to get you information about the sensor board I use.

Serveurperso, 2017-09-19:
5 MP Omnivision 5647

JamesH65, 2017-09-19:
The max analogue gain that sensor can do is 8, which is what I am seeing being used.

Serveurperso, 2017-09-19:
I just want push up the gain when I'm in near night. clearly the sensor is underexposed and I want to see the little CMOS noise at night.

Serveurperso, 2017-09-19:
you say there is a hardware limitation of 8 for the analog gain this is bad:( no way to hack?

JamesH65, 2017-09-19:
As for sensor noise, to see that you would need to turn of the denoise processing - gain is not the only thing that can affect the pixels. There are over 20 stages of processing done to the bayer from the camera before its get displayed/encoded.

AFAIK, 8 is the maximum analogue gain the sensor can support.

Serveurperso, 2017-09-19:
do you know another way to get a hardware accelerated MJPEG stream on raspberry (an PAL analog capture is good for me) ?

Serveurperso, 2017-09-19:
I close this issue after the last test tonight...

JamesH65, 2017-09-19:
Sorry, not sure what you mean by accelerated MJPEG stream.

Serveurperso, 2017-09-19:
raspivid is very fast : do you process by software or get the MJPEG frames from hardware ?

Serveurperso, 2017-09-19:
Look the picture (there is clouds lol) : http://www.serveurperso.com:8080/
[image]
For me this sensor is crappy underexposed.
Even with maximum tuning:(
I must add the dynamic range compensation to high

JamesH65, 2017-09-19:
The entire path from sensor to H264/MJPEG is done in the GPU using HW blocks. The final result is passed to the ARM/Linux userland. The GPU does have software on it to coordinate everything, but that is closed source firmware.

Serveurperso, 2017-09-19:
input_raspicam.so -fps 25 -q 10 -x 720 -y 576 -rot 180 -ex night -ISO 800 -ev 10 -drc high
I don't get the video when I add the -drc option.
I must test with raspivid and if it work I ask the mjpg encoder experimental author.

Serveurperso, 2017-09-19:
versus this one (HAD2 sensor -> analog video downlink -> expensive & own design video receiver -> capture board -> mjpg streamer)

[image]

Serveurperso, 2017-09-19:
And my CCD sensor on very dark :

[image]

sensitivity/gain is "*10 or *100 more OK" than the raspicam...

Serveurperso, 2017-09-19:
And my better camera same light level (near 1 inch CMOS b&w sensor for astronomy)

[image]

6by9, 2017-09-19:
B&W gains you photons hitting the sensor because you remove the colour filters.

The OV5647 has 1.4x1.4um pixels (IMX219 is 1.1x1.1um). Looking at https://www.ptgrey.com/exview-had-ccd-ii-sensor-technology the smallest of those has 3.1. So that's at least 4.9 times as many photons.

And I won't ask about the difference in price....

Use the appropriate tool for your job - that may not be the Pi camera module. Despite the smaller pixel size the IMX219 is typically more sensitive, but not orders of magnitude.

I did start on a patch set that allowed manual control of analog and digital gain, but I can't recall what state it was in. It's not automatic though, so you'd be relying on auto exposure to handle changes in brightness. The use case is too limited to add in extra tuner configs.

Serveurperso, 2017-09-19:
I know... it is a pity that it is not possible to push the automatic gain on the Raspicam, because with the manual light and contrast settings (80 and 90) I can get a descent sensitivity on dark, bug way over exposed on day...
This is an important feature missing. Many cheap webcam can do it and I we clearly see the CMOS working at maximum performance. The pi cam look always very under exploited on night scenes.

But this gain problem is only for video @ fast FPS because slow shutter speed give amazing result on raspicam (I have many PI webcam runing in automatic FPS mode https://www.serveurperso.com/?page=webcam&n=11 doing long exposure )

Serveurperso, 2017-09-19:
For all my PI cam, as objective I use the two first, CS and PCB mount to get a maximum of light :

http://www.framboise314.fr/wp-content/uploads/2014/07/3_types_camera_pi.jpg

but problem is not the sensor sensitivity or optics, it's a too limited automatic maximum gain.
I do this to make a cheap opensource version of my project.

6by9, 2017-09-19:
So actually you haven't purchased a camera module from Raspberry Pi, but you're complaining about it.....

Serveurperso, 2017-09-19:
I have one official but give the way badest result (too narrow angle and very poor sensitivity).

Serveurperso, 2017-09-19:
(and of course there is the same under exposure problem on the official)

Serveurperso, 2017-09-19:
-ex night alone give the best result possible...
An automatic tuning of light contrast (before MJPEG compression) are needed to get the best picture...

JamesH65, 2017-09-19:
Have you tried the brightness and contrast setting from here?

https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

Serveurperso, 2017-09-19:
Yes but it's not automatic...
With light @ 80 and contrast @ 90 I get good performance on night but overexposed the day...

Serveurperso, 2017-09-19:
I'm at home I tested the 6by9 cmdline :
```
raspivid -fps 25 -w 720 -h 576 -cd MJPEG -o /var/www/temp/video.mjpg -set

mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=274/256, B=332/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=278/256, B=333/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=281/256, B=334/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=284/256, B=335/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=287/256, B=336/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=290/256, B=337/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=287/256, B=330/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=284/256, B=323/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=281/256, B=316/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=278/256, B=310/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=276/256, B=304/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=274/256, B=298/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=272/256, B=294/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=270/256, B=290/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=268/256, B=287/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=266/256, B=284/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=264/256, B=281/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=263/256, B=278/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=275/256, B=276/256
mmal: Exposure now 39827, analog gain 2048/256, digital gain 383/256
mmal: AWB R=286/256, B=274/256
```

rodizio1, 2018-02-22:
> So actually you haven't purchased a camera module from Raspberry Pi, but you're complaining about it.....

> I have one official but give the way badest result (too narrow FOV and very poor sensitivity). pinhole are bulshit.

I also buy 3rd-Party cameras. Not because I want to save money, support copy-cats or harm the Raspberry Foundation. Problem is simply, there is no alternative lens for the Raspberry Cams and the FOV of the lens is way too small for my application.

Is there a special reason the cam was designed with a hole-pattern that is exatly one mm off compared to M12 lens mounting holes?

Maybe the Raspberry foundation could consider:
- Making a camera with standard M12 mounting hole spacing so that people can fit any standard M12 lens holder
- Offer some optional M6 lenses with different field-of-view

Serveurperso, 2018-02-22:
For a robot the wide angle / fisheye is 100% mandatory, you can test to move my 3rd-camera here :
http://www.serveurperso.com:8080/
[image]

closed 2018-10-29 by JamesH65