raspberrypi/userland #425: The image rotates to get the wrong image

tpengti, 2017-09-21:
When I set the following parameters, the use of video streams in the form of data back to opencv inside.
-rot, --rotation   90
-w, --width 1944
-h, --height 2592

static void video_buffer_callback(MMAL_PORT_T *port, MMAL_BUFFER_HEADER_T *buffer)
{
	MMAL_BUFFER_HEADER_T *new_buffer;
	RASPIVID_STATE * state = (RASPIVID_STATE *)port->userdata;		
	......
		if (buffer->length)
		{
			int w=state->width;	// get image size
			int h=state->height;
			int pixelSize = state->monochrome ? 1 : 3;
			memcpy(state->dstImage->imageData,buffer->data,w*h*pixelSize);	
        ...........
}
The image shows garbled, should be the data format is wrong, consult the difference between the data, when the image has rotated 90 degrees.

6by9, 2017-09-21:
I'm assuming you're refering to either raspi(still)yuv or raspividyuv, but you don't say so.

The stride of the image buffers is always aligned to a multiple of 32, and the height is aligned to a multiple of 16. That is what is reflected in the format->es->video.[width|height] fields vs the format->es->video.crop.[width|height] values. 
1944 is not a multiple of 32, so you need to treat the buffer as 1952x2592 with the last 8 pixels of each line being just padding.

tpengti, 2017-09-21:
@6by9 Understand, thank you very much

6by9, 2017-09-24:
Please close the issue if you have the explanation you need.

tpengti, 2017-09-25:
@6by9 
Hello, I add a question, IMX219, 800W resolution is length and width are?  (3264*2448)

6by9, 2017-09-25:
Sorry, I don't fully understand what you're asking. 800W resolution?

All formats adopt padding on the stride to a multiple of 32 pixels, and on the height to a multiple of 16 pixels. 3264x2448 is already aligned appropriately (3264 = 102*32, 2448 = 153*16).
Rotation is irrelevant as that is done to fit into the specified image buffer.

JamesH65, 2018-07-02:
Closing this issue as questions answered/issue resolved.

closed 2018-07-02 by JamesH65