raspberrypi/userland #429: Auto exposure issue with timelapse capture

tomsaul, 2017-10-04:
When using the -t <duration> -tl <frequency> options to capture a sequence of images (with or without -bm) the application is not waiting to get an appropriate determination of the exposure to use.  With -bm it simply uses the first value (typically very over exposed), and without it it generally slowly converges on a proper exposure as more frames are taken.  I suspect there should be a sleep (perhaps with the duration indicated by another setting) to allow the exposure to be determined after the camera is activated, but before images are saved.  This is done in some of the other capture methods - but not for timelapse!

6by9, 2017-10-05:
So what value have you specified as duration and frequency?
The code does wait for ```frequency``` milliseconds before taking the first capture.

I've just run ```raspistill -w 640 -h 480 -o foo%d.jpg -t 30000 -tl 2000 -bm, -set```, and it takes captures starting at 2 seconds, which is more than enough time for AE to settle. It always starts at exposure of 10ms and gain x1.0, and in two runs it's adapted to 1.4msec and x1.0 when pointed out the window, or 32.9ms and x1.3 gain looking in the office.

tomsaul, 2017-10-05:
So far anything below about -tl 1000 results in this problem.  The faster the rate, the more pronounced it is.  The point is that this capture should not start before the hardware is ready, so the rate specified should not be a factor.  The other cases in the code seem to add a fixed delay (1000 as I recall) to allow time for this process, but that was skipped on timelapse mode.  Another option would be to change the initial capture to something like `max(<tl interval>, <init delay>)`, where `<init delay>` could be the new setting, or the hard coded value used elsewhere!

tomsaul, 2018-05-07:
And FWIW - the exact same issue occurs with the raspivid program.  The image initially displays overexposed, and slowly corrects.

JamesH65, 2018-05-07:
I would expect value below 1000 to start to become unreliable. The systems need time to start up the camera, receive the first frame, dump it, tyhen start to receive good frames, then get start its AE/AGC loops. That can  take upto 1000ms (more if you are setting large shutter times).

tomsaul, 2018-05-07:
This is the crux of the problem report.  Most other modes of capture have a fixed startup delay to allow this to occur.  The Timelapse mode in raspicam does not, and while I have not looked at the code in raspivid, my guess is that it also is not doing this.  In both cases, all the image control options are using their defaults.

JamesH65, 2018-07-02:
IIRC, there's no real fixed delay, the system will default to the 5000ms before taking the picture but you can change that using -t xxxx. In timelapse mode, the delay before the first image is the frequency, that's just how raspistill works. Think we are starting to hit the limits of what raspistill was designed for - simply as demonstration for people to adapt as they see fit.  It's had lots of stuff bolted on since then, and is starting to creak at the seams.

Waiting for the HW to become ready is not the right approach (from a HW perspective, its ready very quickly)- you actually need to wait for the exposure loops to settle down, and at the moment there is no way of determining that from RaspStill/Vid.

tomsaul, 2018-07-02:
The existing code does include a fixed delay for most of the capture modes.  It just is not present for the timelapse mode.  If there is a better option than raspistill I'd love to hear about it, but short of that adding the same delay (or some better variation of it) would vastly improve the experience.

JamesH65, 2018-07-02:
I'd have to check the code tomorrow but I don't remember putting in a fixed delay. Might have been someone else.

tomsaul, 2018-07-03:
Look at line 1664 in raspistill.c - several of the other cases do not include a similar 'wait to stabilize' delay, but I suspect most of them get this inherently while they wait for some other external action.  My guess is that line 1604 should be something like 
vcos_sleep(max(state->timelapse, 1000));

JamesH65, 2018-07-03:
Ah, that code was me, but 5 years ago, so no wonder I don't remember it. 

That bit of code I don't think is used in timelapse mode, you need to check the line 1604, where for the first frame it waits for the timelapse value - that should have a similar change to the one you proposed. Give it a go, easy enough to compile userland and try out changes, report back if it works.

JamesH65, 2018-07-18:
OK, taking a look at this, seems a simple enough change that won't break too many things, although there is the related #473 which seems to want the opposite! I'll have a think.

PR #476 (2018-07-18): Fixed up initial timelapse delay
When in timelapse mode, the delay before the first shot
was the timelapse value. However, if a low value is set
this does not give enough time for the AE etc to settle.

In addition, the first image was only taken after the
first delay, rather than immediately which is what would
be expected. So a 5minute timelapse delay meant the 1st
image was only taken after 5 minutes.

The patch sets the delay for the first image capture to
be a constant value, 1000ms. So always time for the AE etc
to settle.

Fixes: #429, #473
commit 8b28006c (2018-07-18): Fixed up initial timelapse delay

When in timelapse mode, the delay before the first shot
was the timelapse value. However, if a low value is set
this does not give enough time for the AE etc to settle.

In addition, the first image was only taken after the
first delay, rather than immediately which is what would
be expected. So a 5minute timelapse delay meant the 1st
image was only taken after 5 minutes.

The patch sets the delay for the first image capture to
be a constant value, 1000ms. So always time for the AE etc
to settle.

Fixes: #429, #473
closed 2018-07-18 by pelwell