raspberrypi/userland #431: Raspberry Camera generates only 1 keyframe for h264

kemotz, 2017-10-10:
Hello,

I have a raspberry 1 B+ with camera connected, raspbian 9 stretch.
I am using the h264 stream to feed SIP client - BareSIP- via v4l2_codec module that grabs h264 frames from v4l2 source and passes it to the SIP client.

All works fine, however the raspicam generates only one keyframe and the others are deltas. This leads to issues in case the first frame is dropped leading to no video stream (no subsequent keyframes).

I am trying to use the following command, but it doesn't help:
raspivid -fps 24 -w 858 -h 480 -ih -g 10 -if both

Please let me know how effectively to set the keyframes frequency and how to debug it.

Thanks,
Tomek.

JamesH65, 2017-10-10:
-g should be the option you want. But I think by default, it should be adding one keyframe every second anyway, so not sure why yours seems to be not producing keyframes - it's not something that has been reported elsewhere. How are you checking for them?

6by9, 2017-10-10:
Appears to be working fine for me.

Add a line:
```
printf("Length %d, flags %04X, pts %llu\n", buffer->length, buffer->flags, buffer->pts);
```
to https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVid.c#L1416 and rebuild.
Running ```raspivid -fps 24 -w 858 -h 480 -ih -g 10 -if both -o file.h264``` I'm getting output such as 
```
Buffer length 28, flags 00000024, pts 0
Buffer length 29240, flags 0000000C, pts 632239919
Buffer length 20198, flags 00000004, pts 632283305
Buffer length 8252, flags 00000004, pts 632326691
Buffer length 9010, flags 00000004, pts 632370077
Buffer length 9145, flags 00000004, pts 632413463
Buffer length 9363, flags 00000004, pts 632456849
Buffer length 15215, flags 00000004, pts 632500235
Buffer length 9004, flags 00000004, pts 632543622
Buffer length 8217, flags 00000004, pts 632587007
Buffer length 8961, flags 00000004, pts 632630394
Buffer length 28, flags 00000024, pts 0
Buffer length 29693, flags 0000000C, pts 632673778
Buffer length 9567, flags 00000004, pts 632717166
Buffer length 8420, flags 00000004, pts 632760550
Buffer length 8515, flags 00000004, pts 632803937
Buffer length 8573, flags 00000004, pts 632847322
Buffer length 7926, flags 00000004, pts 632890708
Buffer length 7989, flags 00000004, pts 632934093
Buffer length 8306, flags 00000004, pts 632977479
Buffer length 7615, flags 00000004, pts 633020865
Buffer length 7673, flags 00000004, pts 633064251
Buffer length 28, flags 00000024, pts 0
Buffer length 29828, flags 0000000C, pts 633107639
Buffer length 8847, flags 00000004, pts 633151023
Buffer length 7744, flags 00000004, pts 633194411
...
```
The 28 byte buffer is the SPS (19 bytes) and PPS (8 bytes), correctly flagged with 0x20 for ```MMAL_BUFFER_HEADER_FLAG_CONFIG``` (https://github.com/raspberrypi/userland/blob/master/interface/mmal/mmal_buffer.h#L120), and 0x04 for ```MMAL_BUFFER_HEADER_FLAG_FRAME_END```. The buffer immediately following is an IDR frame, and is correctly flagged with ```MMAL_BUFFER_HEADER_FLAG_KEYFRAME | MMAL_BUFFER_HEADER_FLAG_FRAME_END```.

(0 as the PTS on the header bytes is a little odd. I'm not quite sure what's going on there, but I'm not running a standard firmware)

Failing that, we need to know what kernel and firmware you are running. ```uname -a``` and ```vcgencmd version```.

6by9, 2017-10-10:
> -g should be the option you want. But I think by default, it should be adding one keyframe every second anyway

Default intra-I period is 30 frames, so once a second if running at 30fps, but here we're asking for 24 so every 1.25secs.

kemotz, 2017-10-10:
Gents, thanks for prompt reply!
I am getting the info from the BareSIP directly after finishing the call:
v4l2_codec: encoder stats (keyframes:1, deltaframes:772)
And the behavior seems to confirm it.
```
root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.9.53+ #1040 Fri Oct 6 14:15:15 BST 2017 armv6l GNU/Linux
root@raspberrypi:/home/pi# vcgencmd version
Oct  6 2017 13:59:43 
Copyright (c) 2012 Broadcom
version bde38f02b236b6e1bd16eb0f9b44b35f5a85cb30 (clean) (release)
```
@6by9 , Could you please please confirm that line with "Buffer length 28" indicate a Keyframe?

Thanks,
Tomek.

6by9, 2017-10-10:
> Could you please confirm that lines with "Buffer length 28" indicate Keyframes?

No, flags with 0x20/```MMAL_BUFFER_HEADER_FLAG_CONFIG``` set are codec header bytes. With ```-ih``` the header bytes are always sent before a key frame. For H264 they will be 28 bytes long.
A key frame will have 0x08/```MMAL_BUFFER_HEADER_FLAG_KEYFRAME``` set in the flags field.

Rereading your comments are you actually using V4L2 (bcm2835_v4l2) rather than piping the output of raspivid? That's a totally different situation as camera applications are totally independent (other than that you can't capture images from more than 1 at a time). Your mention of raspivid threw me on that.

If using v4l2, then V4L2 remembers the state of all settings.
You can use v4l2-ctl to query and set them.
```v4l2-ctl --list-ctrls``` will list all the controls and their current settings.

```v4l2-ctl -v width=640,height=480,pixelformat=H264``` will set the current resolution/format to H264 VGA, and then ```v4l2-ctl --stream-mmap=3 --stream-count=100 --stream-to=/dev/null``` will stream for 100 frames and discard the output. You should however see K's being marked to denote keyframes. In testing that I have noted that there is a patch missing from 4.13 and 4.14, but 4.9 should be fine.
Use ```v4l2-ctl --set-ctrl=h264_i_frame_period=10``` to set the intra-I-frame interval to 10 frames. ```v4l2-ctl --set-ctrl=repeat_sequence_header=1``` to request header bytes before every key frame.

kemotz, 2017-10-10:
My understanding was that raspivid writes the parameters to the hardware somehow, which was wrong. 
Using the v4l2-ctl with parameters you have mentioned @6by9 helped out :-) Thanks a lot!

closed 2017-10-10 by kemotz