raspberrypi/userland #432: Double-buffered EGL context, possible with dispmanx?

vanfanel, 2017-10-16:
Hi there,

I know it's unlikely you will be adding new features to the dispmanx interface and brcm EGL/GLES implementation, but I would like to ask if it's possible as things are now.

On DRM/KMS, it's possible to do it, because we first call egl_swap_buffers(), and then we "manually" issue a flip on the buffers of the gbm surface with drmModePageFlip(), so we can do one of these:
-Wait for flip to happen, so we don't loop on the program anymore until it's done: very good for preventing input lag.
-Don't wait for flip to happen *if* we still have free buffers on the gbm surface: the program will loop one more time, thus rendering another frame: one frame of input lag will be introduced, but no CPU time will be "wasted" waiting for flip to happen.

It can be a bit strange when told like this, so you can look at this simple code from RetroArch's gfx_ctx_drm_swap_buffers() function:

`
egl_swap_buffers(&drm->egl);

/* I guess we have to wait for flip to have taken 
* place before another flip can be queued up.
*
* If true, we are still waiting for a flip
* (nonblocking mode, so just drop the frame). */
if (gfx_ctx_drm_wait_flip(drm->interval))
   return;

waiting_for_flip = gfx_ctx_drm_queue_flip();

/* Triple-buffered page flips */
if (video_info->max_swapchain_images >= 3 &&
    gbm_surface_has_free_buffers(g_gbm_surface))
    return;

gfx_ctx_drm_wait_flip(true);
`
Evidently, drmModePageFlip() is called inside gfx_ctx_drm_queue_flip(), and we wait for flip to happen in gfx_ctx_drm_wait_flip().

So, back to the old good dispmanx, all we have is egl_swap_buffers(), and that's it. The underlying implementation will swap buffers in a triple-buffered scheme, with an added frame of delay, and no "finegrain" control at all that I know of. Or maybe there's a way to WAIT for flip to happen before going on?
A mechanism to wait after egl_swap_buffers() has been called until the swap DOES happen: it has to be some blocking call I guess, like in the case of KMS/DRM, so the program won't advance until the requested swap has been done. Is it possible somehow?

Thanks!

popcornmix, 2017-10-17:
By default firmware side EGL is triple buffered and eglSwapBuffers won't block.
If you set env variable ``V3D_DOUBLE_BUFFER=1`` it will double buffer (and block on vsync if running too fast).

You can see this being used by kodi when running on a low memory device:
https://github.com/xbmc/xbmc/blob/master/xbmc/linux/RBP.cpp#L130

vanfanel, 2017-10-17:
@popcornmix : Thanks a lot! I didn't know about that env variable! Very usefull stuff.
But, is a sequence like this supposed to block until next vsync?

```
vc_dispmanx_update_start();
vc_dispmanx_update_submit();
```
Also, no way to do the same as V3D_DOUBLE_BUFFER=1 but with a function call of some kind? For elegance.

popcornmix, 2017-10-18:
```
vc_dispmanx_update_start();
vc_dispmanx_update_submit();
```
will only clock if there is something to do. You'll need to include some element add/modify calls to avoid this being a no-op.

If you just want to block until vsync, you may want to use vc_dispmanx_vsync_callback. See:
https://github.com/xbmc/xbmc/blob/master/xbmc/linux/RBP.cpp#L168

vanfanel, 2017-10-19:
@popcornmix : So, should a simple call like 
`vc_dispmanx_vsync_callback(m_display, NULL, NULL);`
wait for vsync? Even without receiving a callback function pointer?

popcornmix, 2017-10-19:
``setenv("V3D_DOUBLE_BUFFER", "1", 1);`` should be done just once when your app inits before you call any egl functions.
I believe with this set, eglSwapBuffers will block when you are calling it faster than vsync rate. No effect if you are below vsync rate (depending how you handle it, using ``vc_dispmanx_vsync_callback`` may or may not block when running below the vsync rate).

vanfanel, 2017-10-19:
@popcornmix : Well, it seems it doesn't work. 
I am calling `setenv("V3D_DOUBLE_BUFFER", "1", 1);` before any egl call, even before bcm_host_init, and eglSwapBuffers isn't waiting for vsync immediately (double-buffering).

The other aproximation, consisting on waiting after eglSwapBuffers on a condition that is signaled from the vsync callback passed to vc_dispmanx_vsync_callback, DOES work.

So, could you please take a look to be sure that enviroment variable is still taken into account by eglSwapBuffers?

JamesH65, 2018-07-02:
I've checked the code, and the double buffering stuff does seem to be used correctly. String searched for is V3D_DOUBLE_BUFFER, and it does result in 2 buffers being used in eglCreateWindowSurface rather than 3.

closed 2018-10-29 by JamesH65