raspberrypi/userland #434: With a video encoder component, the output callback is only called once for every two input frames

mortie, 2017-10-19:
I have some code which, from my understanding, should be a fairly simple and correct MMAL-based video encoder, but I only receive one output buffer for every two buffers I send to the input port.

My code is based on the example video decoder at https://github.com/raspberrypi/userland/blob/master/interface/mmal/test/examples/example_basic_1.c, and looks essentially like this (but a bit less simple and with error checking; no errors occur, all the mmal-related functions return MMAL_OK):

    void
    Encoder::InputCallback(MMAL_PORT_T* port, MMAL_BUFFER_HEADER_T* buffer) {
        Encoder* ctx = (Encoder*)port->userdata;

        mmal_buffer_header_release(buffer);
        vcos_semaphore_post(&ctx->semaphore_);
    }

    void
    Encoder::OutputCallback(MMAL_PORT_T* port, MMAL_BUFFER_HEADER_T* buffer) {
        Encoder* ctx = (Encoder*)port->userdata;

        mmal_queue_put(ctx->queue_, buffer);
        vcos_semaphore_post(&ctx->semaphore_);
    }

    void Encoder::Init() {
        mmal_component_create(MMAL_COMPONENT_DEFAULT_VIDEO_ENCODER, &component_)
            bcm_host_init();
        vcs_semaphore_create(*semaphore_, "encoder", 1);

        MMAL_ES_FORMAT_T* format_in = component_->input[0]->format;
        format_in->type = MMAL_ES_TYPE_VIDEO;
        format_in->encoding = MMAL_ENCODING_I420;
        format_in->es->video.width = 640;
        format_in->es->video.height = 480;
        format_in->es->video.frame_rate.num = 60
            format_in->es->video.frame_rate.den = 1;
        format_in->es->video.par.num = 1;
        format_in->es->video.par.den = 1;
        format_in->es->video.crop.x = 0;
        format_in->es->video.crop.y = 0;
        format_in->es->video.crop.width = 640;
        format_in->es->video.crop.height = 480;

        mmal_port_format_commit(component_->input[0]);

        MMAL_ES_FORMAT_T* format_out = component_->output[0]->format;
        mmal_format_copy(format_out, format_in);
        format_out->type = MMAL_ES_TYPE_VIDEO;
        format_out->encoding = MMAL_ENCODING_H264;

        component_->input[0]->buffer_num =
            component_->input[0]->buffer_num_min;
        component_->input[0]->buffer_size =
            component_->input[0]->buffer_size_recommended;
        component_->output[0]->buffer_num =
            component_->output[0]->buffer_num_min;
        component_->output[0]->buffer_size =
            component_->output[0]->buffer_size_recommended;

        mmal_port_format_commit(component_->output[0]);

        queue_ = mmal_quee_create();

        component_->input[0]->userdata = (MMAL_PORT_USERDATA_T*)this;
        mmal_port_enable(component_->input[0], InputCallback);
        pool_in_ = mmal_pool_create(
            component_->input[0]->buffer_num,
            component_->input[0]->buffer_size);

        component_->output[0]->userdata = (MMAL_PORT_USERDATA_T*)this;
        mmal_port_enable(component_->output[0], OutputCallback);
        pool_out_ = mmal_pool_create(
            component_->output[0]->buffer_num,
            component_->output[0]->buffer_size);

        mmal_component_enable(component_);
    }

    Encoder::Encode(
            uint8_t* data_y, size_t len_y,
            uint8_t* data_u, size_t len_u,
            uint8_t* data_v, size_t len_v) {
        vcos_semaphore_wait(&semaphore_);

        MMAL_BUFFER_HEADER_T* raw = mmal_queue_get(pool_in_->queue);
        memcpy(raw->data, data_y, len_y);
        memcpy(raw->data + len_y, data_u, len_u);
        memcpy(raw->data + len_y + len_u, data_v, len_v);
        raw->length = len_y + len_u + len_v;

        MMAL_BUFFER_T* encoded;
        MMAL_BUFFER_T* buffer;
        MMAL_BUFFER_T* prevbuf = NULL;
        mmal_port_send_buffer(component_->input[0], raw);
        while ((buffer = mmal_queue_get(queue_)) != NULL) {
            if (prevbuf != NULL)
                mmal_buffer_header_release(prevbuf);
            prevbuf = buffer;
            encoded = buffer;
        }

        // Send empty buffers to the output port
        while ((buffer = mmal_queue_get(pool_out_->queue)) != NULL) {
            mmal_port_send_buffer(component_->output[0], buffer);
        }

        // doSomethingInteresting does something interested with the
        // encoded image
        doSomethingInteresting(encoded);

        if (encoded != NULL) {
            printf("encoded is not null\n");
            mmal_buffer_header_release(buffer);
        } else {
            printf("encoded is null\n");
        }
    }

This code will alternate between printing "encoded is not null" and "encoded is null".

Am I using the MMAL API wrong? Have I misunderstood something about how it works?

6by9, 2017-10-19:
Posting a working example would have been useful as I'm having to recreate your use case. There are typos and missing characters in the code you have posted :-(

You also appear to have assumed that one buffer in equals one buffer out. That is an invalid assumption for several reasons:
- an H264 stream always starts with a set of header bytes (SPS and PPS) that will be used to configure the decoder on playback. Those should be flagged with ```MMAL_BUFFER_HEADER_FLAG_CONFIG``` being set in the flags field.
- An encoded frame can span multiple output buffers if the bitrate allows/requires. A frame is only complete when you get a frame with ```MMAL_BUFFER_HEADER_FLAG_FRAME_END``` set in the flags field.

Several things that look wrong with your code:
- You're not waiting on the semaphore, so you've sent the buffer to the input port and immediately expect a buffer on the output. mmal_queue_get does NOT block, you want mmal_queue_wait if you want a blocking behaviour (or mmal_queue_timedwait if you want a timeout). This is the most likely issue.
- You appear not to have set ```format_out->bitrate```, so potentially the encoder is trying to encode to 0kb/s.
- There are no timestamps set on the input frames, therefore bit rate control will be solely off the configured frame rate.

6by9, 2017-10-19:
Working code based on yours, but done in C rather than C++.

```
#include "bcm_host.h"
#include "interface/mmal/mmal.h"
#include "interface/mmal/mmal_format.h"
#include "interface/mmal/mmal_port.h"
#include "interface/vcos/vcos_semaphore.h"
#include "interface/mmal/util/mmal_default_components.h"
#include <stdio.h>

typedef struct Encoder {
    MMAL_COMPONENT_T *encoder;
    VCOS_SEMAPHORE_T semaphore_;
    MMAL_QUEUE_T *queue_;
} Encoder;

Encoder state;

void InputCallback(MMAL_PORT_T* port, MMAL_BUFFER_HEADER_T* buffer) {
    Encoder* ctx = (Encoder*)port->userdata;

    mmal_buffer_header_release(buffer);
    vcos_semaphore_post(&ctx->semaphore_);
}

void
OutputCallback(MMAL_PORT_T* port, MMAL_BUFFER_HEADER_T* buffer) {
    Encoder* ctx = (Encoder*)port->userdata;

    mmal_queue_put(ctx->queue_, buffer);
    vcos_semaphore_post(&ctx->semaphore_);
}

int main(void)
{
    MMAL_COMPONENT_T *component_;
    MMAL_QUEUE_T *queue_;
    MMAL_POOL_T *pool_in_, *pool_out_;
    int i;

    bcm_host_init();
    mmal_component_create(MMAL_COMPONENT_DEFAULT_VIDEO_ENCODER, &component_);
    state.encoder = component_;
    vcos_semaphore_create(&state.semaphore_, "encoder", 1);

    MMAL_ES_FORMAT_T* format_in = component_->input[0]->format;
    format_in->type = MMAL_ES_TYPE_VIDEO;
    format_in->encoding = MMAL_ENCODING_I420;
    format_in->es->video.width = 640;
    format_in->es->video.height = 480;
    format_in->es->video.frame_rate.num = 60;
    format_in->es->video.frame_rate.den = 1;
    format_in->es->video.par.num = 1;
    format_in->es->video.par.den = 1;
    format_in->es->video.crop.x = 0;
    format_in->es->video.crop.y = 0;
    format_in->es->video.crop.width = 640;
    format_in->es->video.crop.height = 480;

    mmal_port_format_commit(component_->input[0]);

    MMAL_ES_FORMAT_T* format_out = component_->output[0]->format;
    mmal_format_copy(format_out, format_in);
    format_out->type = MMAL_ES_TYPE_VIDEO;
    format_out->encoding = MMAL_ENCODING_H264;
    format_out->bitrate = 1000000;

    component_->input[0]->buffer_num =
        component_->input[0]->buffer_num_min;
    component_->input[0]->buffer_size =
        component_->input[0]->buffer_size_recommended;
    component_->output[0]->buffer_num =
        component_->output[0]->buffer_num_min;
    component_->output[0]->buffer_size =
        component_->output[0]->buffer_size_recommended;

    mmal_port_format_commit(component_->output[0]);

    queue_ = mmal_queue_create();
    state.queue_ = queue_;

    component_->input[0]->userdata = (void*)&state;
    mmal_port_enable(component_->input[0], InputCallback);
    pool_in_ = mmal_pool_create(
        component_->input[0]->buffer_num,
        component_->input[0]->buffer_size);

    component_->output[0]->userdata = (void*)&state;
    mmal_port_enable(component_->output[0], OutputCallback);
    pool_out_ = mmal_pool_create(
        component_->output[0]->buffer_num,
        component_->output[0]->buffer_size);

    mmal_component_enable(component_);

//    uint8_t* data_y, size_t len_y,
//    uint8_t* data_u, size_t len_u,
//    uint8_t* data_v, size_t len_v)

    vcos_semaphore_wait(&state.semaphore_);

    MMAL_BUFFER_HEADER_T* raw = mmal_queue_get(pool_in_->queue);
    //memcpy(raw->data, data_y, len_y);
    memset(raw->data, 0x80, raw->alloc_size);
    //memcpy(raw->data + len_y, data_u, len_u);
    //memcpy(raw->data + len_y + len_u, data_v, len_v);
    raw->length = raw->alloc_size; //len_y + len_u + len_v;

    //MMAL_BUFFER_HEADER_T* encoded;
    MMAL_BUFFER_HEADER_T* buffer;
    //MMAL_BUFFER_HEADER_T* prevbuf = NULL;
    int in_count = 1, out_count = 0;
    printf("Send input buffer %d\n", in_count);
    mmal_port_send_buffer(component_->input[0], raw);

    for (i=0; i < 100 && out_count<10; i++)
    {
        vcos_semaphore_wait(&state.semaphore_);

        if ((buffer = mmal_queue_get(queue_)) != NULL) {
            out_count++;
            printf("New output buffer %d, length %d, flags %04X\n", out_count, buffer->length, buffer->flags);
            mmal_buffer_header_release(buffer);
        }

        // Send empty buffers to the output port
        while ((buffer = mmal_queue_get(pool_out_->queue)) != NULL) {
            mmal_port_send_buffer(component_->output[0], buffer);
        }

        // Requeue input buffers to the input port
        while ((buffer = mmal_queue_get(pool_in_->queue)) != NULL) {
            in_count++;
            printf("Send input buffer %d\n", in_count);
            buffer->length = buffer->alloc_size;
            mmal_port_send_buffer(component_->input[0], buffer);
        }

        // doSomethingInteresting does something interested with the
        // encoded image
        //doSomethingInteresting(encoded);

        //if (encoded != NULL) {
        //    printf("encoded is not null\n");
        //    mmal_buffer_header_release(buffer);
        //} else {
        //    printf("encoded is null\n");
       // }
    }
}
```

I'll add that your memcpy's of the data are dubious as they don't necessarily match with what the GPU is expecting.
For I420 the Y data will be at offset 0 and run for ```format->es->video.width * format->es->video.height``` bytes.
U will be at offset ```format->es->video.width * format->es->video.height``` and ```(format->es->video.width * format->es->video.height) >> 2``` in length.
V will be at offset ```(format->es->video.width * format->es->video.height) + ((format->es->video.width * format->es->video.height) >> 2)``` and ```(format->es->video.width * format->es->video.height) >> 2``` in length.
```length``` must be >= ```((format->es->video.width * format->es->video.height) * 3) >>1```.

6by9, 2017-10-19:
The output I get from my code:
```
Send input buffer 1
Send input buffer 2
New output buffer 1, length 28, flags 0024, pts 0
Send input buffer 3
New output buffer 2, length 99, flags 000C, pts 0
Send input buffer 4
New output buffer 3, length 19, flags 0004, pts 16666
Send input buffer 5
New output buffer 4, length 19, flags 0004, pts 33332
Send input buffer 6
New output buffer 5, length 19, flags 0004, pts 49998
Send input buffer 7
New output buffer 6, length 19, flags 0004, pts 66664
Send input buffer 8
New output buffer 7, length 19, flags 0004, pts 83330
Send input buffer 9
New output buffer 8, length 19, flags 0004, pts 99996
Send input buffer 10
New output buffer 9, length 19, flags 0004, pts 116662
Send input buffer 11
New output buffer 10, length 19, flags 0004, pts 133328
```
Note the two buffers with PTS 0. The first with flags 0x24 (```MMAL_BUFFER_HEADER_FLAG_CONFIG | MMAL_BUFFER_HEADER_FLAG_FRAME_END```), whilst the second has 0x0C (```MMAL_BUFFER_HEADER_FLAG_KEYFRAME | MMAL_BUFFER_HEADER_FLAG_FRAME_END```

Within the component there is an input FIFO of up to normally 3 images, and an output FIFO of normally 2MB, so the output can fall significantly behind the input if you're not consuming the data in a timely manner.

mortie, 2017-10-20:
I'm sorry for wasting your time, I wasn't expecting you to try to run my code, just for it to serve as an illustration of how I was using MMAL, and I wrote it in a bit of a hurry. I see now I should've made a complete example which actually compiles.

Thanks a ton, your code works very well, and with your help I was able to get my code to work too.

closed 2017-10-20 by mortie

lherman-cs, 2020-10-20:
> Working code based on yours, but done in C rather than C++.
> 
> ```
> #include "bcm_host.h"
> #include "interface/mmal/mmal.h"
> #include "interface/mmal/mmal_format.h"
> #include "interface/mmal/mmal_port.h"
> #include "interface/vcos/vcos_semaphore.h"
> #include "interface/mmal/util/mmal_default_components.h"
> #include <stdio.h>
> 
> typedef struct Encoder {
>     MMAL_COMPONENT_T *encoder;
>     VCOS_SEMAPHORE_T semaphore_;
>     MMAL_QUEUE_T *queue_;
> } Encoder;
> 
> Encoder state;
> 
> void InputCallback(MMAL_PORT_T* port, MMAL_BUFFER_HEADER_T* buffer) {
>     Encoder* ctx = (Encoder*)port->userdata;
> 
>     mmal_buffer_header_release(buffer);
>     vcos_semaphore_post(&ctx->semaphore_);
> }
> 
> void
> OutputCallback(MMAL_PORT_T* port, MMAL_BUFFER_HEADER_T* buffer) {
>     Encoder* ctx = (Encoder*)port->userdata;
> 
>     mmal_queue_put(ctx->queue_, buffer);
>     vcos_semaphore_post(&ctx->semaphore_);
> }
> 
> int main(void)
> {
>     MMAL_COMPONENT_T *component_;
>     MMAL_QUEUE_T *queue_;
>     MMAL_POOL_T *pool_in_, *pool_out_;
>     int i;
> 
>     bcm_host_init();
>     mmal_component_create(MMAL_COMPONENT_DEFAULT_VIDEO_ENCODER, &component_);
>     state.encoder = component_;
>     vcos_semaphore_create(&state.semaphore_, "encoder", 1);
> 
>     MMAL_ES_FORMAT_T* format_in = component_->input[0]->format;
>     format_in->type = MMAL_ES_TYPE_VIDEO;
>     format_in->encoding = MMAL_ENCODING_I420;
>     format_in->es->video.width = 640;
>     format_in->es->video.height = 480;
>     format_in->es->video.frame_rate.num = 60;
>     format_in->es->video.frame_rate.den = 1;
>     format_in->es->video.par.num = 1;
>     format_in->es->video.par.den = 1;
>     format_in->es->video.crop.x = 0;
>     format_in->es->video.crop.y = 0;
>     format_in->es->video.crop.width = 640;
>     format_in->es->video.crop.height = 480;
> 
>     mmal_port_format_commit(component_->input[0]);
> 
>     MMAL_ES_FORMAT_T* format_out = component_->output[0]->format;
>     mmal_format_copy(format_out, format_in);
>     format_out->type = MMAL_ES_TYPE_VIDEO;
>     format_out->encoding = MMAL_ENCODING_H264;
>     format_out->bitrate = 1000000;
> 
>     component_->input[0]->buffer_num =
>         component_->input[0]->buffer_num_min;
[... 72 lines cut ...]
>         }
> 
>         // doSomethingInteresting does something interested with the
>         // encoded image
>         //doSomethingInteresting(encoded);
> 
>         //if (encoded != NULL) {
>         //    printf("encoded is not null\n");
>         //    mmal_buffer_header_release(buffer);
>         //} else {
>         //    printf("encoded is null\n");
>        // }
>     }
> }
> ```
> 
> I'll add that your memcpy's of the data are dubious as they don't necessarily match with what the GPU is expecting.
> For I420 the Y data will be at offset 0 and run for `format->es->video.width * format->es->video.height` bytes.
> U will be at offset `format->es->video.width * format->es->video.height` and `(format->es->video.width * format->es->video.height) >> 2` in length.
> V will be at offset `(format->es->video.width * format->es->video.height) + ((format->es->video.width * format->es->video.height) >> 2)` and `(format->es->video.width * format->es->video.height) >> 2` in length.
> `length` must be >= `((format->es->video.width * format->es->video.height) * 3) >>1`.
> 
> _edited to add pts to the data_

@6by9 in this case, what's the recommended way to write raw images to the buffer?