raspberrypi/userland #441: INLINE_MOTION_VECTOR: possible SAD values

jorisvervuurt, 2017-11-20:
The [imv2text example](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/imv_examples/imv2txt.c) uses the `short` to store the SAD value. That would indicate the value could range from -32,768 to 32,767.

Who can tell me what the actual minimum and maximum SAD values are (outputted by `raspivid` using the `-x`)? Thanks in advance!

JamesH65, 2017-12-05:
I've taken a quick look at this - internally the data is moved around without specifying it range. It's basically the output from the CME (coarse motions estimator) which is a 32 bit value, I think two 16 bits coordinates. They are stored in a macroblock sized grid buffer (ie x x y macroblocks * sizeof(uint32_t)).

My guess is that they are in pixel or quarter pixel coordinates, in which case the range is either going to be +- width or width*4, +- height or height *4, or thereabouts. I'll keep looking around.

jorisvervuurt, 2017-12-05:
Thanks for looking into this!
For now, I'm just looking at the minimum and maximum values I'm seeing and guessing the sweet spot. I've seen the value going from 0 to > 1500 so far.

JamesH65, 2017-12-05:
I suspect its is related to the resolution of the image. Divide that up by macroblock size (they are 8x8), to give macroblock resolution. Probably 4x as quarter pixels as mentioned above, so for example,640x480 max value woudl be +- 640/8*4 = +-320x240

But that is a guess.

jorisvervuurt, 2017-12-06:
I indeed think it's related to the resolution of the image, but I'm afraid your calculation isn't correct. I'm using a resolution of 1920x1080 and see a SAD value that ranges from 0 to about 1500. As 1920/8*4 = 960, I think it's too far off from 1500. The mystery continues... 🤣

JamesH65, 2017-12-06:
Like I said, I am guessing, and having trouble finding anyone or any docs
that might explain the exact output from the CME. It may be in actual
pixels rather than sub-macroblock positions.

6by9, 2017-12-06:
It's the best Sum of Absolute Differences (SAD) on the luma samples within the subsampled (8x8) macroblock search.
Therefore the max would be when the best result is switching from a pure luma 0 to pure luma 255, so a difference of 255 * 8*8 = 16320.

The hardware docs say it to output an unsigned value, which makes sense as the absolute difference can never be negative. It's 15 bit as one bit is used as a flag for invalid, so it got squashed into a signed 16 bit value. (It doesn't look like the invalid flag gets masked off, so negative values should be treated as invalid).

jorisvervuurt, 2017-12-06:
Thanks for the explanation!
So, a valid range is from 0 to 16320, no matter the image resolution?

6by9, 2017-12-06:
> So, a valid range is from 0 to 16320, no matter the image resolution?

That's my expectation, yes.
For info, the CME appears to be searching a range of +/-5 macroblocks horizontally, and +/-4 vertically, so that would be the limit on the motion vector size.

JamesH65, 2017-12-06:
Of course, the aim of the compression is the reduce the SAD to a minimum, so the chances of getting any numbers even remotely close to 16320 are vanishingly small.

(And sorry, I've been talking about vectors all along rather than the SAD you originally mentioned)

Closing as answered.

closed 2017-12-06 by JamesH65