raspberrypi/userland #443: H264 encoder component doesn't repeat PPS/SPS NAL units

mortie, 2017-12-07:
Some H264 decoders (notably [Google Chrome's decoder used for WebRTC](https://groups.google.com/forum/#!topic/discuss-webrtc/QbiKSphZlE4)) expects session and picture parameter sets to be sent with every keyframe, but MMAL's encoder component doesn't do that. I don't know what the H264 specification says about the matter, but since some decoders depend on it, it might be worth considering at least adding an option to repeat the PPS and SPS.

Or is there such an option already? I have looked for an option by grepping through the source code, but there might be something I've missed (as the MMAL library doesn't seem to be too thoroughly documented).

6by9, 2017-12-07:
```MMAL_PARAMETER_VIDEO_ENCODE_INLINE_HEADER``` (maps down to ```OMX_IndexParamBrcmVideoAVCInlineHeaderEnable``` if you were using IL).

mortie, 2017-12-07:
Awesome, that's exactly what I was looking for. Thanks.

closed 2017-12-07 by mortie