raspberrypi/userland #444: RaspiStill: Camera v2 (IMX219 sensor) supported?

Mostafa248, 2017-12-24:
I would like to use **raspistill** with the Camera V2 module (IMX219 sensor), but looking at the source code `RaspiStill.c`, it seems configured for the Camera V1 (OV5647 sensor).

Is camera v2 supported by RaspiStill?

6by9, 2017-12-25:
Yes, it is supported.
The code queries the GPU firmware for the capabilities (predominantly resolution) of the sensor. If that fails then it is likely to be due to an ancient firmware, and an ancient firmware will only support OV5647, hence the defaults.

Mostafa248, 2017-12-31:
Wonderful!  I would like to suggest mentioning this in a comment in the source code (so other users with similar doubt can know).

Thank you very much for the prompt response

JamesH65, 2017-12-31:
Not sure any sort of specific comment is needed - all our software is compatible with all our products, unless otherwise  specified. Looking at the source code to see about compatibility seems rather a long winded approach - simply running raspistill (installed as standard) with the camera attached would show that it works.

closed 2017-12-31 by JamesH65