raspberrypi/userland #450: RaspiStill verbose to stderr

ledazul, 2018-01-29:
Documentation of raspistill says that if a filename is not specified, the output is sent to stdout, which is fine.
It does not says where the verbose info is sent. It is actually sent to stderr, wich is unexpected but understandable.

Problem is that the verbose info is disabled if the output is sent to stdout:

https://github.com/raspberrypi/userland/blob/9c42baea7b291308fff9b7d8e5afe28c528dc81d/host_applications/linux/apps/raspicam/RaspiStill.c#L1960-L1966

Since one is sent to stdout and the other to stderr, can verbose really upset the output stream?
An equivalent question is, if verbose is sent to stderr, why is it disabled when a capture is sent to stdout?

6by9, 2018-01-29:
You're quite correct that as all debug output is sent to stderr, it doesn't matter whether the output goes to file or stdout.
Lines 1964/1965 are incorrect and should be removed. Fancy producing a Pull Request for us? (Remove the blank line at 1963 too please).

JamesH65, 2018-01-29:
I think that is a hangover from when all output was sent to stdout, as you
didn't want the verbose stuff being injected in to the data stream.

ledazul, 2018-01-29:
I think I can't produce a pull request, GitHub help says I must have write access to a repository to create a branch and open a pull request.
I am not sure how it's done, I am new here :)

JamesH65, 2018-01-29:
You need to clone the code tree to your own Github, make the changes, then issue a pull request from your repo to the Pi one

ledazul, 2018-01-29:
Thanks, I think I produced the pull request

6by9, 2018-01-29:
Merged. Thanks for your contribution.

closed 2018-01-29 by 6by9