raspberrypi/userland #464: raspivid segmentation fault with latest Stretch

Hermann-SW, 2018-05-02:
pi@raspberrypi3Bplus:~ $ raspivid -md 7 -fps 64 -o test.h264
Too many macroblocks/s: Increasing H264 Level to 4.2
pi@raspberrypi3Bplus:~ $ raspivid -md 7 -fps 65 -o test.h264
mmal: Too many macroblocks/s requested
Segmentation fault
pi@raspberrypi3Bplus:~ $

ethanol100, 2018-05-02:
You are requesting 1920x1080 at 65 fps which is not valid within H264 Level 4.2.
Raspivid could check for out of bound values, but then we would need to lower the fps or the resolution and how should rasapivid decide?

6by9, 2018-05-02:
Very true that it is well outside the capabilities of the hardware, but it also shouldn't segfault.

It should be a 2 minute job with gdb to see where it is throwing the seg fault, but I can't see it immediately, and I haven't got a suitable system to hand.

Hermann-SW, 2018-05-10:
>  but it also shouldn't segfault.
>
That was why I reported.

PR #466 (2018-05-11): RaspiVid: Return an error if the overall encode exceeds level 4.2
The error handling forgot to set an error status, so main continued
on and then seg faulted due to no encoder component.

Fixes #464.
commit 24486446 (2018-05-11): RaspiVid: Return an error if the overall encode exceeds level 4.2

The error handling forgot to set an error status, so main continued
on and then seg faulted due to no encoder component.

Fixes #464.
closed 2018-05-11 by pelwell