raspberrypi/userland #465: raspivid annotation error

tomsaul, 2018-05-08:
When using raspivid and adding an annotation with a background it is not formatting properly.  THe command line I am using is 'raspivid --annotate 1036 --annotateex 16,0xff,0x808000 ...'.  The background is not aligning with the text (see example).

[image]

6by9, 2018-05-11:
Please confirm the version of the firmware you are running (```vcgencmd version```), and the resolution you are encoding at (you've trimmed your command).

There was a bug I recall from ages ago - version control suggests Feb 2015, so it really should be resolved.

Just doing ```raspivid -a 1024 --annotateex 16,0xff,0x808000 -a "09:35:31 05/08/18" -o /dev/null``` which should reproduce your example image is working fine for me (defaults to 1920x1080 for the resolution).

tomsaul, 2018-05-11:
vsgencmd result is 
pi@Cam1:~ $ vcgencmd version
Jul  3 2017 14:20:31
Copyright (c) 2012 Broadcom
version 4139c62f14cafdb7d918a3eaa0dbd68cf434e0d8 (tainted) (release)

If I use a hard coded -a value (i.e. -a "00:00:00 00/00/0000") it is fine, but if I use -a 1036 or -a 1024 -a "%X %D" it exhibits the same offset problem.  It is also fine if I don't use the -ae, but then I can't control the font size used.

6by9, 2018-05-11:
July 2017 is pretty old so worth checking with the latest firmware. Having said that I can't see any recent changes relating to the text overlay.

To track this down I need to know under EXACTLY what conditions it shows up. Please quote your FULL raspivid command, not just snippets. It's going to be some particular combination of width of the overlay text with regard 128 pixel boundaries that is causing the problem. It's strange that hard coded text isn't triggering it as they both go through the same sprintf.

tomsaul, 2018-05-11:
Here are a few of the variations I have tried.  Is there a newer release, and if so how do I update?  I have done the standard apt-get updates (from standard repositories).

This works as expected - text is in the box.
raspivid -mm matrix --annotate 1024 -ae 16,0xff,0x808000 -a "00:00:00 00/00/0000" -n -t 15100 -w 800 -h 600 -fps 5 -o /tmp/$USER-$mac.h264 && MP4Box -new -add /tmp/$USER-$mac.h264:fps=5 
/tmp/$USER-$mac.mp4 && rm /tmp/$USER-$mac.h264 && ./camsend.sh /tmp/$USER-$mac.mp4 $mac.mp4

This fails
raspivid -mm matrix --annotate 1036 --annotateex 16,0xff,0x808000 -n -t 15100 -w 800 -h 600 -fps 5 -o /tmp/$USER-$mac.h264 && MP4Box -new -add /tmp/$USER-$mac.h264:fps=5 /tmp/$USER-$mac.mp4 && rm /tmp/$USER-$mac.h264 && ./camsend.sh /tmp/$USER-$mac.mp4 $mac.mp4

This also fails
raspivid -mm matrix --annotate 1024 -ae 16,0xff,0x808000 -a "%X %D" -n -t 15100 -w 800 -h 600 -fps 5 -o /tmp/$USER-$mac.h264 && MP4Box -new -add /tmp/$USER-$mac.h264:fps=5 
/tmp/$USER-$mac.mp4 && rm /tmp/$USER-$mac.h264 && ./camsend.sh /tmp/$USER-$mac.mp4 $mac.mp4

6by9, 2018-05-11:
Thank you. 800x600 does appear to trip it up, hence why I asked what resolution you were using in my first response. I can investigate next week.

If apt update, apt upgrade isn't getting you a more recent firmware then I surmise you're still on the Jessie release. ```cat /etc/os-release``` to confirm. That is no longer supported as Stretch was released the middle of last year, hence the lack of updates.
The best solution is to update to Stretch so that you get udpates on all packages.
You can use ```sudo rpi-update``` to get the absolute latest firmware and kernel, but please use caution on that as there is the potential for the odd regression creeping in.

tomsaul, 2018-05-11:
Yes - on Jessie

root@Cam1:/home/pi# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

I'll try some variations on image size to see if I can circumvent in the interim.  Not sure if I want to deal with dev versions of OS for this.

tomsaul, 2018-05-11:
FWIW - so far the only sizes I have found where it works properly are 640x480 and full size.  1024x768 does not, 1000x750 does not.

I'm also having issues with a secondary -a not working as expected - for example "raspivid -a 1024 -a "%X" ..." is showing '%X' in the image, not the time.  I'm fairly certain this was working earlier, so I will test further.

6by9, 2018-05-13:
And "%X" by itself has never been supported by itself to include the time/date. You can set the time and/or date flags (4 and 8) and then provide an override on the formatting string, but %X by itself will print those two characters.

6by9, 2018-05-14:
A corner case.
At text size 16 the default time/date string ends up at exactly 128 pixels wide and the splitting between columns code is slightly wrong.

Quickest fix for you is to add ```-a "%X %x "``` to add a single space to the overall string, and that fills in the background correctly. I'll sort out a firmware patch to fix the underlying issue in one of the upcoming releases.

tomsaul, 2018-05-15:
Thanks - and FWIW - it was not clear (to me) from the documentation that the -a 'strftime type formatting' also required that you include the respective numerical mask (-a 4, -a 8, -a 12) to activate/allow it.  In any case, adding some extra padding has done the trick.

6by9, 2018-05-22:
The proper fix should now be merged into the firmware if you use ```sudo rpi-update``` to update.

As usual be aware that this picks up the "testing" branch for both firmware and kernel, so either backup first, do it on an unimportant sdcard, or accept that there is a small chance of regressions.

6by9, 2018-05-24:
I'm happy that this issue is fixed.
Please confirm for your test case, otherwise I'll close the issue in a week.

6by9, 2018-06-12:
Closing as it is understood that the issue is fixed.

closed 2018-06-12 by 6by9

tomsaul, 2020-04-20:
This seems to be fixed in version (commit 06bc6daa0213 Tainted)

6by9, 2020-04-20:
It's a firmware fix, not userland. Any firmware from 18th May 2018 or onwards should have fixed it.