raspberrypi/userland #474: OMX_EmptyThisBuffer failure with OMX_ErrorBadParameter

estobbart, 2018-07-05:
I'm attempting to use only the `audio_render`, I've setup the input port for s16le PCM data, the component transitions state(s) to executing correctly. But when I call OMX_EmptyThisBuffer, it returns OMX_ErrorBadParameter. From the Khronos docs it's not clear why OMX_EmptyThisBuffer would return that. I've verified the parameters to the function, and the component and the buffer header are correct. Could it be a parameter not set correctly on the component? or a value of the buffer header?

6by9, 2018-07-06:
The only logic I can see that returns BadParameter is
```
   if (pBuffer == NULL || pBuffer->nSize != sizeof(OMX_BUFFERHEADERTYPE) ||
       pBuffer->nOffset + pBuffer->nFilledLen > pBuffer->nAllocLen)
      return OMX_ErrorBadParameter;
```
It'd be worth validating those parameters first.
Otherwise if you can provide a simple test case then we can have a look.

estobbart, 2018-07-06:
That was it, I was off by 1 in the `nOffset`. Thank you.

closed 2018-07-06 by estobbart