raspberrypi/userland #480: OMX.broadcom.video_decoder OMX_EventPortSettingsChanged

estobbart, 2018-07-20:
It's widely documented in books, online & in reference implementations to wait for the settings changed event from the decoder before enabling the ports to the rest of the video pipeline. I'm assuming to prevent allocating output/input buffers twice.

Are there other options in the start up sequence to know when this event might occur? I had assumed the PPS & SPS of an H264 stream would trigger it.

Or can the pipeline be preconfigured using the avcC box info to prevent or ignore the event?

popcornmix, 2018-07-23:
@6by9 may know more.
But output format changes in the stream can occur.
These include things like width/height, pixel aspect ratio and framerate (rare when playing video files, but more common in broadcast TV).

6by9, 2018-07-23:
I don't have the firmware source to hand, but my recollection is that if you have already set the output port to the same resolution as the headers will result in then it will just start decoding happily into those buffers. Should at any point it get a set of headers then it will produce a further OMX_EventPortSettingsChanged event and halt decoding until the port has been disabled, port definition updated, and the port reenabled and buffers allocated.
(The V4L2 codec driver under development is using the same underlying component, and when used with GStreamer format changes are not handled, however it does program up the output side correctly ahead of time, and it does then all run).

estobbart, 2018-07-23:
By headers, you mean the PPS & SPS in the stream? or the actual OMX buffer header type?
So for example if it gets an SPS/PPS where the decoder output port isn't configured for that wdith,height/aspect?
Which would mean you could potentially ignore that event if you read those nal's and configured it when they changed?

6by9, 2018-07-23:
Yes, SPS and PPS headers in the stream for H264. The same component handles MJPEG too where there are no explicit headers required (I can't recall the situation for MPEG4, MPEG2, or VC1, which are also all handled).

If the format matches then IIRC it doesn't generate the event at all, because if it does generate the event then it **will** stop decoding until the buffers are reallocated to match.

closed 2018-07-23 by estobbart