raspberrypi/userland #484: Question: Is dtoverlay still active and maintained here?

bclouser, 2018-08-29:
Hello, 

I see that _opensrc/helpers/libfdt_ is a snapshot taken 3 years ago which got me wondering if there were plans afoot to update this to the latest?  I am ultimately wondering about the 'dtoverlay' tool. Is this the best place to get and build 'dtoverlay' or is it maintained elsewhere?

Thanks,
  Ben

pelwell, 2018-08-30:
This is the canonical source repo for the dtoverlay utility. libfdt hasn't changed because this snapshot still does what we want. A rainy-day job would be to pull in the latest version and refactor the dtoverlay library to make more use of the now-standard overlay functions, but it's not high on the list of priorities.

JamesH65, 2018-12-03:
I'm thinking this can be closed as its a question that has been answered?

closed 2018-12-03 by pelwell