raspberrypi/userland #485: raspistill won't accept annotation Y shift larger than 715

vladm, 2018-09-06:
I'm trying to move date/time stamp of picture down via --annotateex parameter, full command line is:

`raspistill -a 12  -w 1367 -h 1024 -ae 32,0xaaaaaa,0x8080ff,1,10,715 -o test.jpg`

the odd problem is it works with Y offset set up to 715. Any values larger than 715 would hang raspistill.

6by9, 2018-09-07:
Preview in raspistill defaults to 1024x768, and the annotation settings are shared between preview/video and stills. I'd guess 715 is ending up pushing the annotation off the end of the image. I thought I'd clipped the values so I wouldn't have expected that to lock up, but I'll check it out.

6by9, 2018-09-07:
_Groan_ A ```lines available > lines_required``` instead of >= fails when lines_required is the frame height and is therefore never exceeded. At least it's an easy fix.

6by9, 2018-09-11:
@vladm The fixed firmware is released via rpi-update. Please test it. 
If you're happy then please close the issue.

vladm, 2018-09-11:
@6by9  upgraded kernel to 1.20180910-1  problem went away - thank you

closed 2018-09-11 by vladm