raspberrypi/userland #487: Native build fails

christianrauch, 2018-09-30:
I am trying to build the userland native on x86_64 via `./buildme --native`.

The build fails with `error: size of array ‘vcos_static_assert’ is negative` messages:

```
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:44:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(EGLConfig) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:45:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(EGLContext) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:46:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(EGLDisplay) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:47:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(EGLSurface) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:48:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(EGLClientBuffer) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:49:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(NativeDisplayType) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:50:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(NativePixmapType) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:51:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(NativeWindowType) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:71:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(GLintptr) == 4);
 ^~~~~~~~~~~~~~~~~~
[...]/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array ‘vcos_static_assert’ is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^
[...]/userland/interface/khronos/common/khrn_client_check_types.h:72:1: note: in expansion of macro ‘vcos_static_assert’
 vcos_static_assert(sizeof(GLsizeiptr) == 4);

```

pelwell, 2018-09-30:
This behaviour is by design.

Although the errors are slightly ugly, the `vcos_static_assert`macro is working as expected. If you look at the `note: in expansion` lines you'll see that all of the expressions evaluate to false on your 64-bit platform; the userland repo is not designed to be built on a platform where integers are 64 bits wide, and these static assertions ensure that doesn't happen.

closed 2018-09-30 by pelwell

christianrauch, 2018-09-30:
I tried to build in an x86  virtual machine but got linking errors against arm libraries.
Is it actually possible to build natively on anything other than arm?

pelwell, 2018-09-30:
Some elements can be built natively for 64-bit - the dtoverlay command, for example - but no effort has been made to make the repo cross-platform.

6by9, 2018-09-30:
Which elements are you expecting to be useful on other platforms?

Most of the stuff in the host_applications directory is connecting to GPU based resources which won't exist on any other platform.
Interface is the IPCs.
Containers should build, but you're probably better off using libavformat.
helpers is dtoverlay and some headers.
The rest is pretty much supporting headers .

christianrauch, 2018-09-30:
I basically tried to build it as part of a larger piece of software. I wanted to avoid setting up cross-compiling for the whole project.
I am aware that the userland software is not useful for any other target and that it can indeed be cross-compiled. But if you want to develop software (on an x86 desktop) that is using userland components, you would need to setup cross-compiling for the whole project which is usually not straight forward.