raspberrypi/userland #490: raspistill suppresses weak signal in astro photography

ogeffert, 2018-10-07:
This is a command to take a 6 second long exposure for
quite some time t.
This way I made a series of photo of cassiopeia in the night sky:
 raspistill -q 100 -t 36000000 -ss 6000000 -sh 0 -co 50 -br 50 -sa 0 -ev +2 -ex night -awb auto -ISO 800 -mm average -n -tl 0 -bm -o %04d.jpg

The colour of the stars seem to correspond to the star classes, whoow !!

But _all_ other fainter stars, do not appear ?

What is wrong with the above settings ?

When i use my Olympus camera for night photos all the faint stars are on the
photos, but here they are missing.

ogeffert, 2018-10-07:
[image]

6by9, 2018-10-08:
Please remember that this is a mobile phone camera costing $25, not a DSLR costing several hundred. Whilst it may have the potential to be used for astro photography, it is not sold specifically for that market.
The physical pixels for capturing photons are small compared to standalone cameras, hence the low light performance is lower.

Your raspistill line has several redundant parameters.
-ss 6000000  and -ISO 800 have locked the shutter speed and sensor gain, so adding -ev, -ex, or -mm are going to do nothing as the auto exposure algorithm has no degrees of freedom to adjust.
-sh 0, -br 50, -sa 0, and -awb auto are all the default values, so little point in including them.

There are a couple of processing stages that may be affecting you.
- First is defective pixel correction which tries to compensate for stuck pixels within the sensor. As your stars are going to be so small in terms of pixel counts it may be misdetecting them as problem pixels.
- Second is denoise. Again the small number of pixels means that it may be removing what are effectively low level blemishes.

In either case, the first step is to add the ```-r``` option to capture the raw Bayer image. 
```
git clone https://github.com/6by9/dcraw.git
sudo apt-get install libjasper-dev libjpeg8-dev gettext liblcms2-dev
cd dcraw
./buildme
```
should download and build the fork of dcraw that supports the Pi JPEG+Raw format.
```./dcraw 0000.jpg``` should process the first file and produce 0000.ppm.

Have a look at that to see whether the detail is in the raw file. If the detail isn't then there is little else you can do as it is a pure case of physics - no photons captured = no star shown.
If the detail is, then you can modify raspistill to disable denoise by setting parameter ```MMAL_PARAMETER_STILLS_DENOISE``` to ```MMAL_FALSE```.
Defective pixel correction is a little more involved and requires ```MMAL_PARAMETER_CAMERA_ISP_BLOCK_OVERRIDE``` to be set to ```~0x80``` (or ```0xFFFFFF7F```) - see https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=175711 for more info on that one.

ogeffert, 2018-10-08:
Thanks for the quick answer !

I acknowledge that this camera was not made for astro photography, but
the outcome - coloured stars - was encouraging, so I will give it a hard try.

Unfortunately this will have to wait until next weekend at home - Raspberry plus camera is there.

I wiil first use -r to get the raw data and look if there is detail,
hopefully there is ! 

Otherwise I will try to compile raspistill without denoising and/or
defective pixel correction.

ogeffert, 2018-10-08:
What would be a recommended command line for raspistill towards the night sky:

        raspistill -r -ss 6000000  and what else ??     

The amount of command line arguments and their connections is not easily understood.
I tried a lot, but the results were not better than the photo shown above.

closed 2018-10-08 by ogeffert

ogeffert, 2018-10-08:
Thanks !!!
So I am prepared for the next night sky session :-)

ogeffert, 2018-10-09:
After building dcraw (on my laptop) I realize that the combination of
"raspistill -r ....."to get the raw Bayer data 
and 
applying dcraw (with some options) to the Bayer data,
might give me what I want, much more detail.

So I will try that ---- before building a somewhat different raspistill.

ogeffert, 2018-10-14:
I used the settings you proposed and added a -r,
this way I got a series of photos of a certain start constellation (Cygnus, Cassiopeia, ...).

With  dcraw -k 33 -W *.jpg I created jpgs
(-k to suppress the most noise, you have to find out yourself, -W to avoid getting a light background)
, which I then "stacked" - like astro people do - to remove noise.

The results are much better than using the same approach for my Olympus camera !!!

###### Thanks a lot for helping, I will continue to make astro photos this way :-) ##########

Maybe I'll have to remove the lens, if I start making photos through my telescope.

hrj, 2018-12-28:
@ogeffert Can you please share a sample of the final stacked and denoised photos? Thanks!

ogeffert, 2018-12-28:
Do not expect anything spectacular -- but you see the colours of the stars and get an an idea of their spectral class.
I use the stellarium software to find out, what it knows about a certain star, let's say in the constellation Cassiopeia == Cas.
I put a picture here of 3 constellations: in the centre Cas and around it Ceph and And.
You have to look at it in _real_ darkness, full screen and some magnification.

ogeffert, 2018-12-28:
Hmm, 13.6 MB is too big to be put here.

ogeffert, 2018-12-28:
Sorry, just Cassiopeia, because of this size limitation
[image]

ogeffert, 2018-12-28:
Do not look at it in the browser. 
Download it and look at it in a picture viewer that gives you a full screen without disturbing white or whatever stripes around it.  This gives somewhat the impression of a dark sky.
=====And switch off the light around you.

hrj, 2018-12-29:
@ogeffert Thank you! I can see magnitude 5 stars in there, such as mu-Cas. Spectral class is indeed discernible for the brighter stars.

If you want to capture dimmer stars, it is possible with the RaPi camera if you can tolerate some noise. I think I have managed to capture mag 8 stars with it. I am not able find the images right now, but if I find them, will post them here.

I am also going to try the options mentioned in this thread. The last time I tried (few years back), this stuff was not much documented.

Clear skies!