raspberrypi/userland #494: Needless delay beetwen first and second frame in burst-time-lapse mode

spink-al, 2018-10-29:
https://github.com/raspberrypi/userland/blob/27b4eaa6c8e66a65eddd4d424e724b58e32f7d6e/host_applications/linux/apps/raspicam/RaspiStill.c#L2303
Above line is causing additional needless delay before capturing of 2nd frame in burst-time-lapse mode.

Examples with semi-manual settings: 
1. 5s exposure:
`raspistill --nopreview -v -md 3 -ex off -ss 5000000 -ag 1.0 -dg 1.0 -vf -hf -w 3280 -h 1080 -awb off -awbg 1.0,1.0 -bm -q 100 -tl 0 -t 0 -o work%05d.jpg`
1st frame after ~40s 
2nd frame **20s** after 1st frame
3rd frame 5s after 2nd frame
(...)
.
After changing line 2303 to:
`if (state.burstCaptureMode && frame==0) `
1st frame after ~40s 
2nd frame **5s** after 1st frame 
3rd frame 5s after 2nd frame
(...)

2. 10s exposure:
`raspistill --nopreview -v -md 3 -ex off -ss 10000000 -ag 1.0 -dg 1.0 -vf -hf -w 3280 -h 1080 -awb off -awbg 1.0,1.0 -bm -q 100 -tl 0 -t 0 -o work%05d.jpg`
1st frame after ~80s 
2nd frame **40s** after 1st frame
3rd frame 10s after 2nd frame
(...)
.
After changing line 2303 to:
`if (state.burstCaptureMode && frame==0) `
1st frame after ~80s 
2nd frame **10s** after 1st frame 
3rd frame 10s after 2nd frame
(...)

6by9, 2018-10-29:
That code stinks :-o 
frame can get overridden in so many places that any form of comparison is going to fail.
- use ```-fs 1``` on the command line to start the file numbers at 1.
- use ```-dt``` to store the date/time in the filename.
- use ```-ts``` to store the timestamp in the filename

Setting ```MMAL_PARAMETER_CAMERA_BURST_CAPTURE``` to 1 (should be ```MMAL_TRUE```) repeatedly is valid, so it should be OK to reduce the test down to 
```if (state.burstCaptureMode)```

Could you test that and raise a pull request if it works?

spink-al, 2018-10-29:
Isn't `frame` some internal counter? oO

My current working build contains only `if (state.burstCaptureMode)` without `&& frame==` condition. 
It gives the same outcome with no-delay between 1st and 2nd frame.

I'm kinda desparate to catch some more needless delays, especially before 1st frame. 
So I'll give myself some time before any "pull requests" in case I spot something else.

6by9, 2018-10-29:
> Isn't frame some internal counter?

Sort of. It gets reset to alternate values in several places, hence my comment.
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L2178
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L2192
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L2204

If you're specifiying ALL the exposure/AWB parameters, then you can probably hack the code not to enable the preview output at all and just ask for captures off the stills port. I couldn't swear it works, but it'd be worth a punt.

JamesH65, 2018-10-29:
I wonder why it was done like that in the first place. Don't think it was
me, but memory isn't what it used to be....

spink-al, 2018-10-29:
> If you're specifiying ALL the exposure/AWB parameters, then you can probably hack the code not to enable the preview output at all and just ask for captures off the stills port. I couldn't swear it works, but it'd be worth a punt.

I'll try to do it this way - all parameters are predefined.

spink-al, 2018-10-29:
https://github.com/raspberrypi/userland/blob/27b4eaa6c8e66a65eddd4d424e724b58e32f7d6e/host_applications/linux/apps/raspicam/RaspiStill.c#L2123
Well, @6by9 gave me very pointy clue. 
Deleting line 2123 got me first frame after 33sec in burst mode with 10sec exposures - big improvement comparing to latter 80sec. Similarly 18s compared to 40s with 5sec exposures.

Have to test it now in longer run, especially with shutter times <500000µs when `-ex off` is causing problems.

spink-al, 2018-10-29:
Yeah, that was little off-topic, but maybe someone like me who look here find it useful. 
Or it could be disabled by a commandline switch.
Anyway thank you for pointing in right direction.

JamesH65, 2018-12-03:
Can this be clsoed? I believe the fix has been merged.

closed 2018-12-03 by spink-al